EmuELEC/EmuELEC #12: Kodi Reboots a lot

paolot0rr3s, 2019-01-07:
###Background
Kodi reboots a lot, even goes to safemode a red version of kodi.
###Description
Device is MXIII 4k s912 16gig 1gig ethernet
Iog file attached
[kodi.log](https://github.com/shantigilbert/CoreELEC/files/2732140/kodi.log)

shantigilbert, 2019-01-10:
does it just crash after starting? or are you watching something ?

paolot0rr3s, 2019-01-10:
Not even watching.  Just in idle.  I also updated to the new version. 2.0.1 i got from your projects. The Sx095re.  Its the same.  I will give you the log file tom. Or anything else you want to request?

shantigilbert, 2019-01-10:
Did you do a fresh install? I don't have a S912 device to test, so I can only fully test the S905 version and that does not have that problem :(

paolot0rr3s, 2019-01-10:
It was a fresh install of the both.  The prerelease and the sx095re 2.0. I installed the 2.0.1 via tar update. That is what was available.

shantigilbert, 2019-01-10:
I am compiling a new version and i will upload asap so maybe you can test it out.

paolot0rr3s, 2019-01-10:
Awesome. I will provide logs and figure out whats up on my end after fresh installing your latest compile.  I will wait for that.

paolot0rr3s, 2019-01-10:
First of all amazing version, like the splash screen a lot.  It is working well now.  I think it is a problem with the ouya controller that I am trying to connect.  This worked for me with the ScottElec, but the problem with that version is emulationstation does not work.  I think this is more of a development issue with the kodi controller and Leia.  I will still attach the kodi.log if you are interested.
[kodi_crashlog_20190110163241.log](https://github.com/shantigilbert/CoreELEC/files/2746271/kodi_crashlog_20190110163241.log)
[kodi.log](https://github.com/shantigilbert/CoreELEC/files/2746272/kodi.log)
two files are attached here.

ouya controller works well with emulationstation.

shantigilbert, 2019-01-10:
Thanks :) 

So the problem is mostly with the Ouya controller? if its not connected then kodi doesn't crash? 
I am not familiar with the Ouya controller, is it Bluetooth? wired?

paolot0rr3s, 2019-01-10:
Oh its blue tooth.  What development build of kodi are you using?

shantigilbert, 2019-01-10:
Can you do me a favor and check if the controller works when you use CoreELEC by itself? 

https://github.com/CoreELEC/CoreELEC/releases/tag/8.95.7

paolot0rr3s, 2019-01-10:
Hi @shantigilbert ,

I might have jumped the gun reporting to you.  It still crashes and goes to a kodi safe mode.  Even without the ouya controller.  I simply was scraping my video library then it hanged and did not continue.  Playing movies it crashed as well 20 seconds in.
[kodi_crashlog_20190110201528.log](https://github.com/shantigilbert/CoreELEC/files/2746943/kodi_crashlog_20190110201528.log)
[kodi_crashlog_20190110195427.log](https://github.com/shantigilbert/CoreELEC/files/2746897/kodi_crashlog_20190110195427.log)
 I have the whole kodi-crashed folder that it made if you want me to email it to you.  10mb though.

shantigilbert, 2019-01-10:
Ok, can you try the CoreELEC I linked, see if its something with the base on S912 devices?

paolot0rr3s, 2019-01-11:
ok will do.  I just need to backup this one setup.

paolot0rr3s, 2019-01-12:
I have an update.  the coreelec 8.95.7 works very well.

shantigilbert, 2019-01-12:
Even with the ouya controller?

shantigilbert, 2019-01-13:
I will keep investigating about this issue as I have no idea and can't really test it without a proper S912 box

shantigilbert, 2019-01-18:
@paolot0rr3s I can't reproduce this on a clean install on a S912 device, Kodi does not crash. I have to admit tho, I don't have any local movies, so not sure if when you scan the movies is where it fails, does scanning also crash Official CoreELEC? 

I just release a Kodi Addon that basically adds Sx05RE to official CoreELEC so maybe you can try that?

paolot0rr3s, 2019-01-18:
No it doesn't.

paolot0rr3s, 2019-01-27:
> And the point is? Like I said, I can't reproduce it on the device. "I
> can't reproduce this on a clean install on a S912 device"

Might be something to do with 4k? I have a 4k tv.  I will test every
posibility let me know what you want to do and I will look into it for you.

shantigilbert, 2019-01-27:
Well the best thing to test is if the Official CoreELEC crashes while doing a video scan or just regular use, I don't use Kodi that much to be able to test it properly, but there has been no other reports of people having problems with the build :/ another option is to install the official CoreELEC for S912 and install the Sx05RE Kodi addon and see if that works for you? 

https://github.com/shantigilbert/CoreELEC/releases/tag/addon_v2.2

shantigilbert, 2019-01-30:
@paolot0rr3s can you try the latest release and see if the problem still persists? 

https://github.com/shantigilbert/CoreELEC/releases/tag/v2.2

shantigilbert, 2019-02-02:
I've been using the S912 for the past few hours and I found something that might be your problem.

if I used the default dtb, I get no problems, but I have no ethernet. once I used the one I thought was the "correct" one I had Ethernet but Kodi crashed a LOT.

Now after I found and used the correct dtb, I had Ethernet and 0 crashes, been using YouTube and watching some videos and movies on a USB stick for the past few hours with no problem.

have you tested another DTB from the /devices_tree directory in the sdcard?

shantigilbert, 2019-02-02:
I think I know what it is... will  do a test and get back at you.

paolot0rr3s, 2019-02-02:
I use the 2g 1gig ethernet one.

paolot0rr3s, 2019-02-02:
The addon has problems with the bluetooth controller not detected

paolot0rr3s, 2019-02-03:
I used all the possible device trees inside the folder.  Still the same thing.  soft restarts kodi randomly.  What I find is the coreelec itself does not hang so much.  But the add-on version you supplied us does not recognise any controller that is bluetooth (the only one I have)

shantigilbert, 2019-02-03:
I found the issue and will release a fix shortly :)

shantigilbert, 2019-02-03:
The new 2.2.1 should fix this issue :)

paolot0rr3s, 2019-02-04:
Yes! Its fixed.  Only thing that got lost is its ability to see samba
drives in the network.  I manually mounted it the terminal as an easy fix.

Cheers!
Paolo

shantigilbert, 2019-02-04:
That is extremely weird as nothing was changed regarding that, BUT maybe some of the Kodi changes broke that. Anyways the original issue is fixed, so I am closing this.

closed 2019-02-04 by shantigilbert