EmuELEC/EmuELEC #14: cannot connect Coreelec repository on V2.1

xiaole51, 2019-01-22:
cannot connect Coreelec repository.
The other two can be connected
I use img is Sx05RE-S905.arm-2.1-S905.img.gz

shantigilbert, 2019-01-22:
Log? error?

xiaole51, 2019-01-22:
it kodi.log

`08:00:45.277 T:3720344432   ERROR: Traceback (most recent call last):
                                              File "/home/core/sx05re/CoreELEC/build.Sx05RE-S905.arm-2.1/CoreELEC-settings-70994c7fe88be17c7af02ef1de380e02df541ac1/.install_pkg/usr/share/kodi/addons/service.coreelec.settings/resources/lib/modules/updates.py", line 514, in check_updates_v2
                                              File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
                                              File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
                                            TypeError: expected string or buffer
22:16:19.330 T:3776656240  NOTICE: AddOnLog: PVR IPTV Archive Client: EPG Loaded.
22:27:58.209 T:3835405168  NOTICE: Previous line repeats 2 times.
22:27:58.209 T:3835405168   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.tvmosaic/resources/fanart.png
22:27:58.623 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.touchscreen/resources/fanart.png
22:27:59.398 T:4018135920   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.system.tinc/resources/fanart.png
22:28:00.034 T:3634688880   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.system.syncthing/resources/fanart.png
22:28:01.805 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.openvfd/fanart.jpg
22:28:01.973 T:3835405168   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.net-snmp/resources/fanart.png
22:28:03.055 T:4018135920   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.multimedia.mpd/resources/fanart.png
22:28:03.134 T:3634688880   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.minidlna/resources/fanart.png
22:28:05.849 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.softcam.oscam/service.softcam.oscam-9.0.106.zip
22:28:07.499 T:3977241456   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 403
22:28:07.499 T:3977241456   ERROR: CCurlFile::Open failed with code 403 for https://addons.coreelec.org/9.0/Amlogic/arm/service.softcam.oscam/service.softcam.oscam-9.0.106.zip:
22:28:07.503 T:3977241456   ERROR: CAddonInstallJob[service.softcam.oscam]: failed to download special://home/addons/packages/service.softcam.oscam-9.0.106.zip
22:28:18.671 T:3835405168   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/driver.dvb.sundtek-mediatv/resources/fanart.png
22:28:19.051 T:4018135920   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.system.tinc/resources/fanart.png
22:28:19.593 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/driver.steamcontroller/resources/fanart.png
22:28:20.471 T:3634688880   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.system.syncthing/resources/fanart.png
22:28:22.745 T:4018135920   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.tvmosaic/resources/fanart.png
22:28:24.108 T:3835405168   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.touchscreen/resources/fanart.png
22:28:39.418 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.locale/resources/fanart.png
22:28:39.828 T:3835405168   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.mariadb/resources/fanart.png
22:28:41.769 T:4018135920   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.minidlna/resources/fanart.png
22:28:42.496 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.locale/resources/fanart.png
22:28:42.817 T:3634688880   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.multimedia.mpd/resources/fanart.png
22:28:56.118 T:3634688880   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.locale/service.locale-9.0.102.zip
22:28:57.682 T:3634688880   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 403
22:28:57.682 T:3634688880   ERROR: CCurlFile::Open failed with code 403 for https://addons.coreelec.org/9.0/Amlogic/arm/service.locale/service.locale-9.0.102.zip:
[... cut ...]
02:08:32.546 T:3977241456   ERROR: CCurlFile::Stat - Failed: HTTP response code said error(22) for https://addons.coreelec.org/9.0/Amlogic/arm/service.softcam.oscam/service.softcam.oscam-9.0.106.zip
02:08:34.123 T:3977241456   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 403
02:08:34.125 T:3977241456   ERROR: CCurlFile::Open failed with code 403 for https://addons.coreelec.org/9.0/Amlogic/arm/service.softcam.oscam/service.softcam.oscam-9.0.106.zip:
02:08:34.125 T:3977241456   ERROR: CAddonInstallJob[service.softcam.oscam]: failed to download special://home/addons/packages/service.softcam.oscam-9.0.106.zip`

shantigilbert, 2019-01-22:
This should be fixed, can you try ?

xiaole51, 2019-01-23:
Try again today can become connected.
It should be my network problem.
Sorry, this is not issue.

shantigilbert, 2019-01-23:
no problem :)

closed 2019-01-23 by shantigilbert