EmuELEC/EmuELEC #15: External roms device format issue

wdmomoxx, 2019-01-24:
I tested the S905 external roms, roms can be recognized under FAT/NTSF, but a type of simulator like GB/GBC/GBA can't enter the game.

shantigilbert, 2019-01-24:
Sorry I don't understand the issue... 

can you SSH to your box and send me the sx05re.log?

wdmomoxx, 2019-01-25:
Usb device is fat32 format
the sx05re.log is this.

```
Sx05RE Run Log
First parameter: GBC
Second Parameter: /storage/roms/gb/俄罗斯方块.zip
Run Command is:
/usr/bin/retroarch -L /tmp/cores/gambatte_libretro.so /storage/roms/gb/俄罗斯方块.zip
Emulator Output is:
Wrong data-format, corrupt or unsupported ROM.
```
If the usb device changes to ext4 format, it works fine.

Sorroy for my poor english.

shantigilbert, 2019-01-25:
It seems to me to be a problem with the characters used "俄罗斯方块" I wonder what it gets converted too? do you have any ROM with English characters that you can use to test?

wdmomoxx, 2019-01-26:
I found the problem, not the character. I will unpack the zip and put it into the fat32 device. The file name has not changed and it works fine. I don't understand why there is such a difference under fat32 and ext4.

shantigilbert, 2019-01-27:
Sorry I have no idea, it is clearly a character problem, but since I don't use any of those characters, I can't reproduce nor know why its happening.  Feel free to reopen the issue if you find more information, but as for now I have no idea how to help

closed 2019-01-27 by shantigilbert