EmuELEC/EmuELEC #16: From Kodi to emulationstation

paolot0rr3s, 2019-02-04:
First, thank you for amazing work.

How did you create a submenu in power button to go back to emulationstation? I installed a build and it deleted some settings from the kodi.

shantigilbert, 2019-02-04:
That menu was replaced a long time ago as it depended on the skin that was used, but you can use the included addon to return back to emulation station. Unless I am misunderstanding your question

paolot0rr3s, 2019-02-04:
Oh I see, it depended on the default skin.  Where could I find the addon?
Is this the onescript.sx05re.launcher-2.2-S912.zip
<https://github.com/shantigilbert/CoreELEC/releases/download/v2.2/script.sx05re.launcher-2.2-S912.zip>
?

shantigilbert, 2019-02-04:
The addon is already included in the latest release, just go into the addon
section in kodi and you should see it

paolot0rr3s, 2019-02-04:
Oh I think the Build installation deleted it.  Would that zip file help?  I will try to install it with that.

shantigilbert, 2019-02-04:
No, that addon should. Be included is just a launcher, the other addon will
probably not work

shantigilbert, 2019-02-05:
> Oh yeah its still there. you can close it. emulationstation addon, I found
> it. With kodi you have lots of options for builds. You can research on it.

Sorry I don't think I follow what you are saying but I am glad you found it
and it works. :)

closed 2019-02-05 by shantigilbert