EmuELEC/EmuELEC #17: TODO

shantigilbert, 2019-03-27:
- [ ] Try to port Bluetooth script from Retropie 
- [ ] Create a configuration for each Retroarch core
- [ ] Try to add more control to BGM 
- [ ] Port more RetroPie scripts
- [ ] Try to figure out the mixer so I can play several audio sources at the same time.

SumavisionQ5, 2019-04-10:
About "Create a configuration for each Retroarch core",you can do as what I am doing:
in sx05reRunEmu.sh,add this line:
`RUNTHIS='/usr/bin/retroarch -L /tmp/cores/${EMU}_libretro.so **--config** /storage/.config/retroarch/rertoarch.cfg "$2"'`

escalade, 2019-04-25:
RetroArch has built in wifi/connmanctl support: 

https://github.com/escalade/LibreELEC.tv/blob/le9/packages/escalade/libretro/retroarch/patches/retroarch-05-wifidriver.patch
https://github.com/escalade/LibreELEC.tv/blob/le9/packages/escalade/libretro/retroarch/package.mk#L110

shantigilbert, 2019-04-25:
@escalade thanks for the comment, that is how I've been doing it for the past few years.

Since I removed Kodi I've been looking for something that is almost self contained in Emulationstation  and I wanted something alternate, that was perhaps easier? (although the Retroarch method is very easy) so I adapted a script https://github.com/shantigilbert/emuELEC/blob/EmuELEC/packages/sx05re/tools/retropie/retropie/scripts/modules/wifi.sh that uses a file in /flash/wifi.txt so that you can set it up on the computer or in /storage/.config/wifi.txt

escalade, 2019-04-25:
Cool, that seems useful :)

shantigilbert, 2019-04-25:
Yeah I assume it can be simplified even more by cleaning the script, but I am happy enough with it right now.

escalade, 2019-04-25:
You're sourcing some stuff from /retropie in the start, is that needed or have you made it self contained? How are you running this? Looks like it's set up with joy2key? Gamepad and virtual keyboard somehow?

shantigilbert, 2019-04-25:
Retropie scripts run on a terminal on the RPI, unfortunately we don't have that on LE/CE (or I could not find ANY info/help on that issue) so I included fbterm https://github.com/shantigilbert/emuELEC/tree/EmuELEC/packages/sx05re/tools/fbterm and run them all in this fb-virtual terminal, that way they display on the TV as a regular terminal. I also included bash (replacing the implementation from busybox), dialog and terminfo so that the scripts could be run with minimal changes. 

The only reason I source the stuff from /retropie is because I've been too lazy/busy to clean what I really need to run those scripts, so right now I source everything until I get the time to clean them up. 
But the idea is to eventually clean all those scripts and only use what is needed, I'll get there someday :P 

joy2key as in RetroPie, is called before any script so that I can control them without a keyboard, you can still use a real keyboard and in some cases it is required for example when you run the Midnight Commander script. 

There is still a lot of work to be done on this part, for example the terminal screen does not show any ANSI characters, can't seem to find a solution for that, so borders and stuff like that show as ASCII  characters.

shantigilbert, 2019-12-16:
List has been completed, closing.

closed 2019-12-16 by shantigilbert