EmuELEC/EmuELEC #20: S912 defaults back to 32bpp color even if forced to 16bpp

shantigilbert, 2019-05-01:
Some emulators (Reicast, OpenBOR, Drastic, Retroarch shaders) on the S912 need to have color depth set to 16bpp or else only a black screen is shown. When running any emulator the color depth switches to 

```
mode "1920x1080-0"
	# D: 0.000 MHz, H: 0.000 kHz, V: 0.000 Hz
	geometry 1920 1080 1920 3240 32
	timings 0 0 0 0 0 0 0
	accel false
	rgba 8/16,8/8,8/0,0/0
endmode
```
# What I tried

I've tried forcing the 16bpp before running an emulator by setting 
``` 
fbset -g 1920 1080 1920 2160 16
```
and also 
```
fbset -g 1920 1080 1920 3240 16
```

Also I tried forcing 16BPP on the FB drivers directly, the system boots to 16BPP but when I run an emulator it switches back to 32BPP.

Also tried patching the dtb (this is the only place I found any mention of 3240) but the same thing happens

```
--- a/arch/arm64/boot/dts/amlogic/gxm_q200_2g.dts
+++ b/arch/arm64/boot/dts/amlogic/gxm_q200_2g.dts
@@ -161,7 +161,7 @@
 		mem_size = <0x01851000 0x00100000>; /* fb0/fb1 memory size */
 		display_mode_default = "1080p60hz";
 		scale_mode = <1>; /** 0:VPU free scale 1:OSD free scale 2:OSD super scale */
-		display_size_default = <1920 1080 1920 3240 32>; //1920*1080*4*3 = 0x17BB000
+		display_size_default = <1920 1080 1920 3240 16>; //1920*1080*4*3 = 0x17BB000
 		logo_addr = "0x7d851000"; /*ion base + fb0 memory size for uboot logo osd1*/
 	};
 	ge2d {
```

I am out of ideas, so I am asking for help on this issue, hopefully someone can help.

kszaq, 2019-05-01:
I suspect that fb bpp is hardcoded in `/system/build.prop`. You can try running this:
```
cp /system/build.prop /storage/build.prop
sed -i 's/32/16/g' /storage/build.prop
mount -o bind /storage/build.prop /system/build.prop
```
after this restart all applications using fb. Alternatively you can run first 2 lines, add 3rd line to `autostart.sh` and reboot.

shantigilbert, 2019-05-01:
Thanks! that actually worked...kinda but now the apps that did work at 32bpp don't work anymore but this will get me in the right direction, I guess I could set it up before running each program that needs it and restore it after it ends.

kszaq, 2019-05-01:
If you have a bind-mounted `build.prop` in `/storage`, you can simply change the value in this file before running an app.

shantigilbert, 2019-05-01:
Thank you, I really appreciate it! This worked for one app (Reicast) the others are weird, it seems they need 32bpp to load but 16bpp to run. (If I start them at 16 in build.pro they crash but if I start them at 32BPP on build.prop they run no problem, but only a black screen and if I set BPP to 16 while they are running I can see image but it all scrambled)

closed 2019-05-13 by shantigilbert