EmuELEC/EmuELEC #21: How to add custom script to emuelec?

farhanito, 2019-05-09:
Is it possible to add custom script to "RetroPie collection" tab?

shantigilbert, 2019-05-09:
On the standalone yes, I am not sure on the add-on

But you can see the path in es_systems.cfg is "/emuelec/scripts/modules" 
So just drop whatever .sh file you want there and it should work.

farhanito, 2019-05-09:
Thanks.

Yes, I am using the standalone. 
The script shows up, but not working. 

Here's what I am trying to do.. 
https://discourse.coreelec.org/t/sx05re-retro-emulation-kodi-addon-for-coreelec-released-for-testing/3203/433?u=farhanito

Script works as expected when executed from SSH.

shantigilbert, 2019-05-09:
What happens when you select it in ES? 
Is there any info on /storage/.emulationstation/es_log.txt 

what happens if you use this script instead?

```
#!/bin/bash

source /emuelec/scripts/env.sh
source "$scriptdir/scriptmodules/supplementary/esthemes.sh"
rp_registerAllModules

joy2keyStart

mount -o rw,remount /flash
mv /flash/boot.ini /flash/boot.ini.emu
read -n 1 -s -r -p "Press any key to reboot"
reboot
```

farhanito, 2019-05-09:
Ah, sorry.. It was Permission issue.

And your script is perfect. 

Thanks again.

closed 2019-05-09 by farhanito

shantigilbert, 2019-05-09:
Also, just to be safe, you might want to add a "sync" command before rebooting.

shantigilbert, 2019-05-29:
Custom scripts should now go into "/emuelec/scripts/custom_scripts/"