EmuELEC/EmuELEC #28: Merge linux upstream HID input patch

BetaXOi, 2019-06-20:
https://git.kernel.org/pub/scm/linux/kernel/git/dtor/input.git/commit/?id=6ce901eb61aa30ba8565c62049ee80c90728ef14

current status in master branch:
amlogic-3.10) use LibreELEC/linux-amlogic 95ba9d626c0fce672caa296f5911ab9190881642 **Fixed**
amlogic-3.14) use CoreELEC/linux-amlogic f1bb00821950b0f5cab1b8a820c697ba9fbd3183 **NOT Fixed**
amlogic-4.9) use CoreELEC/linux-amlogic 6f607ec78af4fc755f41a7dd7e73ad6542f9143b **Fixed**

In my case, i wrote a shell to inject input_event to event device of my gamepad when catch a special input_event triggered by a button pressed and hold, the kernel without the patch will auto report a input_event of button release, but it works fine if kernel be patched.

Maybe we should merge this patch to `packages/linux/patches/amlogic-3.14`

closed 2019-06-21 by shantigilbert

BetaXOi, 2019-06-22:
I have a [LeWGP-201](https://usermanual.wiki/Le-Shi-Zhi-Xin-Electronic-Technology/LEWGP-201) gamepad, it will create two event device, one will be used for remote control on Android, and another be used for other, looks like below:

> Sx05RE:~ # ll /dev/input/by-id   
> total 0
> lrwxrwxrwx    1 root     root             9 Jan  1 08:12 usb-MY-POWER_LeWGP-201-event-if01 -> ../event5
> lrwxrwxrwx    1 root     root             9 Jan  1 08:12 usb-MY-POWER_LeWGP-201-event-joystick -> ../event4
> lrwxrwxrwx    1 root     root             6 Jan  1 08:12 usb-MY-POWER_LeWGP-201-joystick -> ../js1
> Sx05RE:~ # ll /dev/input/by-path/
> total 0
> lrwxrwxrwx    1 root     root             9 Jan  1 08:00 platform-c8100580.meson-ir-event -> ../event6
> lrwxrwxrwx    1 root     root             9 Jun 22  2018 platform-dwc2_a-usb-0:1:1.0-event-joystick -> ../event4
> lrwxrwxrwx    1 root     root             6 Jun 22  2018 platform-dwc2_a-usb-0:1:1.0-joystick -> ../js1
> lrwxrwxrwx    1 root     root             9 Jun 22  2018 platform-dwc2_a-usb-0:1:1.1-event -> ../event5
> lrwxrwxrwx    1 root     root             9 Jun 22  2018 platform-gpio_keypad.47-event -> ../event0

The gamepad has no `select` button but `back` button for Android, so i'm trying to remap `back` to `select`, and remap `power` to `select`+`start`.
I did some test, injecting input_event of BTN_A pressed and holded to `event-joystick` , and input_event of release after 3 seconds, see: https://github.com/BetaXOi/EmuELEC/commit/ce9a38c9b418c5d751268de2164a72004adc67f0
But i got a wired result, input_event of release will auto report by kernel:

> Event: time 1560437407.276170, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1560437407.276170, type 1 (Key), code 304 (BtnA), value 1
> Event: time 1560437407.276170, -------------- Report Sync ------------
> Event: time 1560437407.277813, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1560437407.277813, type 1 (Key), code 304 (BtnA), value 0
> Event: time 1560437407.277813, -------------- Report Sync ------------
> 
> 
> Event: time 1560437410.276373, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1560437410.276373, -------------- Report Sync ------------

BetaXOi, 2019-06-22:
My two LeWGP-201 work fine with my custom kernel patched and `remap` binary program:
https://github.com/BetaXOi/EmuELEC/commit/73807261f5ca2cde41e140fa797d9b314bb519ee

I add two line in autostart.sh
```
/storage/remap /dev/input/by-path/platform-dwc2_a-usb-0:1:1.1-event /dev/input/by-path/platform-dwc2_a-usb-0:1:1.0-event-joystick &
/storage/remap /dev/input/by-path/platform-dwc2_b-usb-0:1:1.1-event /dev/input/by-path/platform-dwc2_b-usb-0:1:1.0-event-joystick &
```

shantigilbert, 2019-06-22:
Thanks! 
could this remap program work to send a command on certain key combinations? like to kill a running application? maybe we an use to quit all emulators with a key-combo

BetaXOi, 2019-06-22:
Yes, we can monitor input_event of event device, and do something we want.
But i don't know how to be suitable for all devices, my source code of `remap` is only suitable for my `LeWGP-201`

shantigilbert, 2019-06-22:
shouldn't all evdev devices work?

BetaXOi, 2019-06-22:
I have no other gamepad to test, i don't know whether the `ScanCode` is same for all gamepad.

> Event: time 1420070488.321441, type 4 (Misc), code 4 (ScanCode), **value 90001**
> Event: time 1420070488.321441, type 1 (Key), code 304 (BtnA), value 1
> Event: time 1420070488.321441, -------------- Report Sync ------------
> Event: time 1420070488.433432, type 4 (Misc), code 4 (ScanCode), **value 90001**
> Event: time 1420070488.433432, type 1 (Key), code 304 (BtnA), value 0
> Event: time 1420070488.433432, -------------- Report Sync ------------
> Event: time 1420070489.553487, type 4 (Misc), code 4 (ScanCode), **value 90002**
> Event: time 1420070489.553487, type 1 (Key), code 305 (BtnB), value 1
> Event: time 1420070489.553487, -------------- Report Sync ------------
> Event: time 1420070489.649473, type 4 (Misc), code 4 (ScanCode), **value 90002**
> Event: time 1420070489.649473, type 1 (Key), code 305 (BtnB), value 0
> Event: time 1420070489.649473, -------------- Report Sync ------------

shantigilbert, 2019-06-22:
how can I test?

shantigilbert, 2019-06-22:
I guess they are not, but the key code seems to be the same 
```
Event: time 1561182439.110425, type 1 (Key), code 304 (BtnA), value 0
Event: time 1561182439.110425, -------------- Report Sync ------------
Event: time 1561182439.626431, type 1 (Key), code 305 (BtnB), value 1
Event: time 1561182439.626431, -------------- Report Sync ------------
```

BetaXOi, 2019-06-22:
compile [evtest.c](https://github.com/BetaXOi/EmuELEC/blob/ce9a38c9b418c5d751268de2164a72004adc67f0/tools/test/evtest.c) and run `./evtest <evdev>` catch button info you want.
Then implement the logic code in [remap](https://github.com/BetaXOi/EmuELEC/blob/HID-fix/tools/test/remap.c)

shantigilbert, 2019-06-22:
my other controller does give scancode 

```
Event: time 1561182667.940376, type 4 (Misc), code 4 (ScanCode), value 90001
Event: time 1561182667.940376, type 1 (Key), code 304 (BtnA), value 1
Event: time 1561182667.940376, -------------- Report Sync ------------
Event: time 1561182668.086607, type 4 (Misc), code 4 (ScanCode), value 90001
Event: time 1561182668.086607, type 1 (Key), code 304 (BtnA), value 0
Event: time 1561182668.086607, -------------- Report Sync ------------
Event: time 1561182668.367918, type 4 (Misc), code 4 (ScanCode), value 90002
Event: time 1561182668.367918, type 1 (Key), code 305 (BtnB), value 1
Event: time 1561182668.367918, -------------- Report Sync ------------
Event: time 1561182668.514142, type 4 (Misc), code 4 (ScanCode), value 90002
Event: time 1561182668.514142, type 1 (Key), code 305 (BtnB), value 0
Event: time 1561182668.514142, -------------- Report Sync ------------
```

so it seems they are maybe the same scancodes and keycodes?

BetaXOi, 2019-06-22:
I have no idea, but i think maybe it's feasible just use `Key` and `Value` without `ScanCode` even

shantigilbert, 2019-06-22:
it would be cool to send SIGTERM to a running program with a key combination

BetaXOi, 2019-06-22:
Yes, i will write a `evkill` demo later, maybe tonight.

BetaXOi, 2019-06-23:
https://github.com/BetaXOi/EmuELEC/commit/ec1b19d56447703faf0892082c18ab7bd6191b9d

shantigilbert, 2019-06-23:
Thanks! I made a package with all the tools, you see anything you would like to change ? 

https://github.com/shantigilbert/EmuELEC/commit/0bf6245ec2b3fa2f4a7cf87acd2b49826ca4f942

shantigilbert, 2019-06-23:
I tried evkill and it works perfectly! now all the emulators that don't have instal quit can be closed with the gamepad! 

I do have a small request, not sure how possible it is (I know close to nothing on this) could it be possible to instead of /dev/input/eventx we could use /dev/input/jsx ? This will make it easier to set-up 

thanks again!

BetaXOi, 2019-06-24:
Yes,it's possible, i will improve the code

shantigilbert, 2019-06-24:
Thank you! I made a few changes, it does not print anything and it exits after the command is called.

BetaXOi, 2019-06-24:
please review PR #30

zang74, 2019-06-25:
Does this patch mean that volume can be controlled outside of RetroArch?

shantigilbert, 2019-06-25:
@zang74 no, but I am curious as to why you would think that, maybe I missed something? 

This patch (the original HID input) is a patch that fixes some particular issues with certain keyboards. 
The second one is a small program that @BetaXOi made for quitting emulators with a game pad key-combo, because some emulators needed keyboard to quit, not anymore :)

zang74, 2019-06-25:
It's in regards to the second one. If it's possible to map a key combo to quitting an emulator, is it not possible to map another to system audio volume?

shantigilbert, 2019-06-25:
Not with this current program, it is made solely to quit a running program (in this case an emulator).

PR #29 (2019-06-21): Merger linux upstream HID-input patch
HID: input: fix confusion on conflicting mappings
ref: https://git.kernel.org/pub/scm/linux/kernel/git/dtor/input.git/commit/?id=6ce901eb61aa30ba8565c62049ee80c90728ef14

fix #28