EmuELEC/EmuELEC #32: Compile addon error

xiaole51, 2019-07-22:
Compile addon error
Http://sources.libreelec.tv/mirror/capsimg/capsimg-067db4c.tar.gz
can not download.
Libreelec has removed the capsimg directory.
This file has been downloaded at https://codeload.github.com/FrodeSolheim/capsimg/tar.gz/067db4c
Why download again on the libecel website?
I tried editing /EmuELEC/packages/sx05re/tools/capsimg/package.mk
But there is no address about LibELEC.
How to skip download this file from LibELEC?

`
--2019-07-22 22:06:05--  https://github.com/FrodeSolheim/capsimg/archive/067db4c.tar.gz
Resolving github.com (github.com)... 140.82.113.4
Connecting to github.com (github.com)|140.82.113.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/FrodeSolheim/capsimg/tar.gz/067db4c [following]
--2019-07-22 22:06:06--  https://codeload.github.com/FrodeSolheim/capsimg/tar.gz/067db4c
Resolving codeload.github.com (codeload.github.com)... 140.82.113.9
Connecting to codeload.github.com (codeload.github.com)|140.82.113.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 127300 (124K) [application/x-gzip]
Saving to: '/home/gonggy/EmuELEC/sources/capsimg/capsimg-067db4c.tar.gz'

          /home/gon   0%[                    ]       0  --.-KB/s               
         /home/gong  15%[==>                 ]  19.44K  96.9KB/s               
        /home/gongg  41%[=======>            ]  51.55K   128KB/s               
       /home/gonggy  96%[==================> ] 119.88K   199KB/s               
/home/gonggy/EmuELE 100%[===================>] 124.32K   205KB/s    in 0.6s    

2019-07-22 22:06:08 (205 KB/s) - '/home/gonggy/EmuELEC/sources/capsimg/capsimg-067db4c.tar.gz' saved [127300/127300]

    [1;31mWARNING[0m      Incorrect checksum calculated on downloaded file: got cfa9acf876254426781925366af67d70d392542d90d6367288d1ab162fc1a7e8 wanted 0576a49a23c70b0022d423d8ae4691c8d7004ceed1527b971f2eba0fc799c0af
--2019-07-22 22:06:08--  http://sources.libreelec.tv/mirror/capsimg/capsimg-067db4c.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-07-22 22:06:08 ERROR 404: Not Found.

Cant't get capsimg sources : https://github.com/FrodeSolheim/capsimg/archive/067db4c.tar.gz
Try later!
`

shantigilbert, 2019-07-22:
Check this commit:

https://github.com/shantigilbert/EmuELEC/commit/471d58aae502b25a217205c167e376c715b647e9

shantigilbert, 2019-07-22:
I also just pushed an update https://github.com/shantigilbert/EmuELEC/commit/701150ac4778665af2d91b38002d8a39add7f064

xiaole51, 2019-07-23:
The same error happened again.

--2019-07-23 18:16:49--  https://github.com/munt/munt/archive/939cc98.tar.gz
Resolving github.com (github.com)... 192.30.253.113
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/munt/munt/tar.gz/939cc98 [following]
--2019-07-23 18:16:51--  https://codeload.github.com/munt/munt/tar.gz/939cc98
Resolving codeload.github.com (codeload.github.com)... 140.82.113.9
Connecting to codeload.github.com (codeload.github.com)|140.82.113.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 418151 (408K) [application/x-gzip]
Saving to: '/home/gonggy/EmuELEC/sources/munt/munt-939cc98.tar.gz'

          /home/gon   0%[                    ]       0  --.-KB/s
         /home/gong   4%[                    ]  19.44K  96.9KB/s
        /home/gongg  12%[=>                  ]  50.21K   125KB/s
       /home/gonggy  28%[====>               ] 115.87K   193KB/s
      /home/gonggy/  67%[============>       ] 277.07K   345KB/s
/home/gonggy/EmuELE 100%[===================>] 408.35K   412KB/s    in 1.0s

2019-07-23 18:16:53 (412 KB/s) - '/home/gonggy/EmuELEC/sources/munt/munt-939cc98.tar.gz' saved [418151/418151]

    [1;31mWARNING[0m      Incorrect checksum calculated on downloaded file: got 0d73b8dd2c9543b9fddb17ce9c6ca6729b4ef124009704a9891ef030fbfd9c35 wanted fb6577e09bd373875464c54ea64f572c97b642da00c84a9d47af9db2c00d1dba
--2019-07-23 18:16:53--  http://sources.libreelec.tv/mirror/munt/munt-939cc98.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-07-23 18:16:54 ERROR 404: Not Found.

shantigilbert, 2019-07-23:
git pull again it should be fixed

xiaole51, 2019-07-24:
again error..

`--2019-07-24 12:20:50--  https://github.com/RetroPie/sixaxis//archive/f53b0ca28c35ebd71b54190f33eadcb8c3267186.tar.gz
Resolving github.com (github.com)... 140.82.113.3
Connecting to github.com (github.com)|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/RetroPie/sixaxis//tar.gz/f53b0ca28c35ebd71b54190f33eadcb8c3267186 [following]
--2019-07-24 12:20:51--  https://codeload.github.com/RetroPie/sixaxis//tar.gz/f53b0ca28c35ebd71b54190f33eadcb8c3267186
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 400 Bad Request
2019-07-24 12:20:52 ERROR 400: Bad Request.

--2019-07-24 12:20:52--  http://sources.libreelec.tv/mirror/sixaxis/sixaxis-f53b0ca28c35ebd71b54190f33eadcb8c3267186.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-07-24 12:20:53 ERROR 404: Not Found.

Cant't get sixaxis sources : https://github.com/RetroPie/sixaxis//archive/f53b0ca28c35ebd71b54190f33eadcb8c3267186.tar.gz
Try later!
`

shantigilbert, 2019-07-24:
This is something on your end as it works fine here

```
GET      sixaxis (archive)
--2019-07-23 23:36:05--  https://github.com/RetroPie/sixaxis//archive/f53b0ca28c35ebd71b54190f33eadcb8c3267186.tar.gz
Resolving github.com (github.com)... 192.30.253.113
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/RetroPie/sixaxis/tar.gz/f53b0ca28c35ebd71b54190f33eadcb8c3267186 [following]
--2019-07-23 23:36:06--  https://codeload.github.com/RetroPie/sixaxis/tar.gz/f53b0ca28c35ebd71b54190f33eadcb8c3267186
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9905 (9.7K) [application/x-gzip]
Saving to: '/mnt/36adaab0-d029-4e57-95d8-ade947075094/CoreELEC/sources/sixaxis/sixaxis-f53b0ca28c35ebd71b54190f33eadcb8c3267186.tar.gz'
```
Delete the source file in `sources/sixaxis` then try again

xiaole51, 2019-07-24:
I am modifying this file to be passed.

../EmuELEC/packages/sx05re/tools/sixaxis/package.mk
PKG_SITE="https://github.com/RetroPie/sixaxis/" >> PKG_SITE="https://github.com/RetroPie/sixaxis"

More than a symbol "/"

shantigilbert, 2019-07-24:
Thanks, I changed the package as well. I will close this issue now.

closed 2019-07-24 by shantigilbert