EmuELEC/EmuELEC #34: ld.gold: error: cannot find -lpthreads

hzio, 2019-07-25:
I got the following error message while building the image,  here are my os and compiler Version:
- Ubuntu 18.04.2 LTS
- cmake version 3.10.2

**OUTPUT ERROR MESSAGE**
```
Run Build Command:"/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/bin/gmake" "cmTC_1e271/fast"
/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/bin/gmake -f CMakeFiles/cmTC_1e271.dir/build.make CMakeFiles/cmTC_1e271.dir/build
gmake[1]: Entering directory '/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/PPSSPPSDL-e8aad38e772c527db9e3653e4697c288336210b4/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1e271.dir/CheckFunctionExists.c.o
/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/bin/armv8a-libreelec-linux-gnueabi-gcc   -march=armv8-a+crc -mtune=cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mcpu=cortex-a53 -mfloat-abi=hard -mfpu=neon-fp-armv8 -fomit-frame-pointer -Wall -pipe -O3 -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_1e271.dir/CheckFunctionExists.c.o   -c /home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/share/cmake-3.13/Modules/CheckFunctionExists.c
Linking C executable cmTC_1e271
/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1e271.dir/link.txt --verbose=1
/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/bin/armv8a-libreelec-linux-gnueabi-gcc -march=armv8-a+crc -mtune=cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mcpu=cortex-a53 -mfloat-abi=hard -mfpu=neon-fp-armv8 -fomit-frame-pointer -Wall -pipe -O3 -DCHECK_FUNCTION_EXISTS=pthread_create  -march=armv8-a+crc -mtune=cortex-a53 -Wl,--as-needed -fuse-ld=gold  CMakeFiles/cmTC_1e271.dir/CheckFunctionExists.c.o  -o cmTC_1e271 -lpthreads
/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/toolchain/lib/gcc/armv8a-libreelec-linux-gnueabi/8.2.0/../../../../armv8a-libreelec-linux-gnueabi/bin/ld.gold: error: cannot find -lpthreads
CMakeFiles/cmTC_1e271.dir/CheckFunctionExists.c.o:CheckFunctionExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_1e271.dir/build.make:87: cmTC_1e271] Error 1
gmake[1]: Leaving directory '/home/hunter/EmuELEC/build.EmuELEC-Amlogic.arm-2.5.4/PPSSPPSDL-e8aad38e772c527db9e3653e4697c288336210b4/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:121: cmTC_1e271/fast] Error 2
```

Hope to get your help!

shantigilbert, 2019-07-25:
What command are you using to create  the image?

hzio, 2019-07-25:
Command in the index page README.md  @shantigilbert 
```
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image
```
Please help, thanks!

hzio, 2019-07-25:
@shantigilbert  Which platform did you test building image successfully?

shantigilbert, 2019-07-25:
Try with:

```PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image_st```

shantigilbert, 2019-07-25:
also maybe try building the new branch `EmuELEC-CE.9.2`

hzio, 2019-07-25:
What's the difference between branches EmuELEC and EmuELEC.9.2 ?

shantigilbert, 2019-07-25:
emuelec is the old one based on CE 9.0 and Emuelec-CE.9.2 is based on CE 9.2 with newer packages

hzio, 2019-07-25:
@shantigilbert After switching to branch EmuELEC.9.2 and  a long time building, I got the same error as well :(

xiaole51, 2019-07-25:
> @shantigilbert切换到分支EmuELEC.9.2和长时间建设后，我也得到了同样的错误:(

Information should include a log.
This file is at ~/EmuELEC/emuelec-kodi_20190725_193949.log
Dates are randomly generated.

shantigilbert, 2019-07-25:
@hzio it is definitely something you are missing on your build computer, but I am not sure what it could be. did you made any changes to the source at all? 

Are you sure you switched to the new branch? can you try dissabling PPSSPPSDL in emuelec/package.mk and see if you get any further ? 

@xiaole51 that log file is only for the addon, he is trying to build the image.

shantigilbert, 2019-07-25:
also, try `sudo apt-get install libpthread-stubs0-dev`

hzio, 2019-07-26:
@hzio 
1. I DID switch to the branch EmuELEC.9.2 and then build.
2. I didn't make any changes to the source code. 
3. Build log files
- [CMakeError.log](https://pastebin.com/c7B0wWLa)
- [CMakeOutput.log](https://pastebin.com/g23Sstdw)

hzio, 2019-07-26:
Build successfully!
After disabling PPSSPPSDL and changing PKG_URL in file tools/soundfont-generaluser/package.mk to my private ftp url, it works fine eventually !!!

But I'm really confused about PPSSPPSDL building error and gonna find a solution about it.

shantigilbert, 2019-07-26:
Did you try this? sudo apt-get install libpthread-stubs0-dev

shantigilbert, 2019-07-26:
Also maybe you could open a pr with the packages you updated?

hzio, 2019-07-28:
> Did you try this? `sudo apt-get install libpthread-stubs0-dev`

Yes, this package had already been installed before starting the build.

It seems a bug of cmake [link here](https://gitlab.kitware.com/cmake/cmake/issues/16540), what's the version of cmake on your os?

hzio, 2019-07-29:
@shantigilbert Is your Ubuntu server edition or desktop edition?

shantigilbert, 2019-07-29:
I need to check what version of cmake I have. I use desktop version

hzio, 2019-07-29:
@shantigilbert  I changed my compiling computer from Ubuntu server edition to desktop edition, everything works well now :D

shantigilbert, 2019-07-29:
I am glad you got it sorted out! I am going to close this issue now, feel free to open another if you have any other problems

closed 2019-07-29 by shantigilbert