EmuELEC/EmuELEC #35: Compile addon CE.9.2 error.

xiaole51, 2019-07-25:
The last consolidation phase error.
Already generated directory../EmuELEC/retroarch_work/2019-07-25_192812
But did not generate the directory /usr/config

`
skipped build.EmuELEC-Amlogic.arm-devel/mpg123-compat-1.25.10/.install_pkg
skipped build.EmuELEC-Amlogic.arm-devel/SDL-1.2.15/.install_pkg
skipped build.EmuELEC-Amlogic.arm-devel/SDL_net-1.2.8/.install_pkg
cp: Unable to get '/home/gonggy/EmuELEC/retroarch_work/2019-07-25_192812/usr/config' stat: No such file or directory`

shantigilbert, 2019-07-25:
You should see why its skipping those packages, the error is because one of those is not being compiled. can you post the complete log on pastebin.com?

xiaole51, 2019-07-26:
logs does not seem to more detailed information.

logs:
https://pastebin.com/hX8KkNrV

ssh output:
https://pastebin.com/fmCEsAR7

shantigilbert, 2019-07-26:
I think you are going to need to build the image first.

shantigilbert, 2019-07-26:
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image

shantigilbert, 2019-07-26:
Or build just those skipping packages firsts

xiaole51, 2019-07-26:
I have successfully built img file.
But to build addon still the same error.

I try build skipping packages firsts,but nothing had happened.
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC scripts/build retroarch

shantigilbert, 2019-07-26:
It still skip the packages even if the image is fully built???

xiaole51, 2019-07-26:
Yes, it has been completely constructed image.
After compiling addon, the same error occurs.

shantigilbert, 2019-07-26:
It seems its having problem finding the path for some reason, but I don't know why

shantigilbert, 2019-08-02:
Did you ever found what the problem was ?

xiaole51, 2019-08-02:
Yes, I found the reason.

The problem caused by the path error.
bulid path is:            /EmuELEC/build.EmuELEC-Amlogic.arm-9.1-devel/
Skipped error path : /EmuELEC/build.EmuELEC-Amlogic.arm-devel/

I tried editing the emuelec-addon.sh file to be passed.
```
#BUILD_SUBDIR="build.${DISTRO}-${PROJECT}.${ARCH}-${VERSION}"
BUILD_SUBDIR="build.${DISTRO}-${PROJECT}.${ARCH}-9.1-devel"
```

This value is wrong.
`[ -z "$VERSION" ] && VERSION=$(cat $SCRIPT_DIR/distributions/$DISTRO/version | grep LIBREELEC_VERSION | grep -oP '"\K[^"\047]+(?=["\047]) ')`

Zip file has been successfully generated.
However, I found that the addon size has become smaller.
script.emuelec.Amlogic.launcher-2.5.4-Amlogic.zip 225M
script.emuelec.Amlogic.launcher-devel-Amlogic.zip 170M (CE9.2)
I will test after installation.

xiaole51, 2019-08-02:
I have installed test, but missing lot of files
Emulationstation does not work.

Libfreeimage.so.3 => not found
libSDL2-2.0.so.0 => not found
Libpthread.so.0 => /usr/lib/libpthread.so.0 (0xf77b9000)
Libasound.so.2 => /usr/lib/libasound.so.2 (0xf771f000)
libGLESv1_CM.so.1 => /usr/lib/libGLESv1_CM.so.1 (0xf7606000)
Libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0xf757b000)
Libcurl.so.4 => /usr/lib/libcurl.so.4 (0xf7510000)
Libvlc.so.5 => not found

shantigilbert, 2019-08-02:
The size of the zip should be less because I removed some unneeded files, as for this 

```
[ -z "$VERSION" ] && VERSION=$(cat $SCRIPT_DIR/distributions/$DISTRO/version | grep LIBREELEC_VERSION | grep -oP '"\K[^"\047]+(?=["\047]) ')
```
You are right, this seems to be the problem, I don't really use "devel" and I forgot about it, will fix it ASAP. 

But in the meantime, instead of changing this 

```
BUILD_SUBDIR="build.${DISTRO}-${PROJECT}.${ARCH}-9.1-devel"
```
Change this 

```
[ -z "$VERSION" ] && VERSION="devel"
```

shantigilbert, 2019-08-02:
This commit should fix it, can you try it ? 

https://github.com/shantigilbert/EmuELEC/commit/c57df7f39b145fe84706768dc945972744da76bf

xiaole51, 2019-08-02:
I git pull this commit and build.
Emulationstation does not work, missing Dependent library.

`ldd /storage/.kodi/addons/script.emuelec.Amlogic.launcher/bin/emulationstation
	libfreeimage.so.3 => not found
	libSDL2-2.0.so.0 => not found
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0xf7578000)
	libasound.so.2 => /usr/lib/libasound.so.2 (0xf74de000)
	libGLESv1_CM.so.1 => /usr/lib/libGLESv1_CM.so.1 (0xf73c5000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0xf733a000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0xf72cf000)
	libvlc.so.5 => not found
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xf718a000)
	libm.so.6 => /usr/lib/libm.so.6 (0xf710a000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0xf70dd000)
	libc.so.6 => /usr/lib/libc.so.6 (0xf6fa1000)
	/lib/ld-linux-armhf.so.3 => /usr/lib/ld-linux-armhf.so.3 (0xaaee0000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0xf6f8e000)
	librt.so.1 => /usr/lib/librt.so.1 (0xf6f77000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0xf6f5b000)
	libz.so.1 => /usr/lib/libz.so.1 (0xf6f48000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0xf6f1c000)
	libssl.so.1.0.0 => /usr/lib/libssl.so.1.0.0 (0xf6ec8000)
	libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0xf6d53000)
`

shantigilbert, 2019-08-02:
ldd emulationstation is going to show all those libraries missing, because the LD_LIBRARY_PATH is not set, check the file `ee_env.sh`

shantigilbert, 2019-08-07:
Did you figure this one out?

shantigilbert, 2019-08-09:
No reply so I am closing this, feel free to reopen if you still have problems

closed 2019-08-09 by shantigilbert