EmuELEC/EmuELEC #37: Can't connect to XBOX ONE bluetooth controller

shulthz, 2019-08-05:
Can't connect to XBOX ONE bluetooth controller with 'bluetoothctl'.

`Failed to connect: org.bluez.Error.Failed`

shantigilbert, 2019-08-06:
Please check the wiki for Bluetooth instructions. I have one question how did you manage to boot up the phincomm? #27 is having problems with the same device.

shulthz, 2019-08-06:
> 
> 
> Please check the wiki for Bluetooth instructions. I have one question how did you manage to boot up the phicomm? #27 is having problems with the same device.

I think he can get the answer from below：
https://right.com.cn/forum/thread-323198-1-1.html

I followed the wiki for Bluetooth instructions，but failled.

shantigilbert, 2019-10-28:
Can you try with the new version? on the main menu under controllers you should see an option to connect to Bluetooth

shantigilbert, 2019-11-07:
No reply for 10 days, closing issue. Feel free to reopen it again if the problem is still there

closed 2019-11-07 by shantigilbert

shulthz, 2019-11-14:
It's very hard to download the the new release in China, i will try it when dowload it~~!

commit 4b77821f (2021-11-03): liberation-fonts-ttf: update to 2.1.5

update 2.1.2 (2020-12-21) to 2.1.5 (2021-09-30)
release notes:
- https://github.com/liberationfonts/liberation-fonts/releases/tag/2.1.3
- https://github.com/liberationfonts/liberation-fonts/releases/tag/2.1.4
- https://github.com/liberationfonts/liberation-fonts/releases/tag/2.1.5

* Thu Sep 30 2021 Vishal Vijayraghavan <vishalvijayraghavan@gmail.com>
- Releasing liberation-fonts 2.1.5 version
- Resolves #40: More fixes to Superscript and subscript Numbers missing issues
-- fixed inconsistent weight, missing glyphs and GSUB issues

* Tue May 04 2021 Visha [... cut]