EmuELEC/EmuELEC #39: evremap problem

liangsdu, 2019-08-25:
using
 `/emuelec/bin/evremap /dev/input/by-id/usb-MY-POWER_LeWGP-201-event-if01 /dev/input/by-id/usb-MY-POWER_LeWGP-201-event-joystick &`
test with 
`evtest /dev/input/event2`
result is :

> Event: time 1566720565.427202, type 4 (Misc), code 4 (ScanCode), value 9000c
> Event: time 1566720565.427202, type 1 (Key), code 315 (BtnStart), value 1
> Event: time 1566720565.427202, -------------- Report Sync ------------
> Event: time 1566720565.547205, type 4 (Misc), code 4 (ScanCode), value 9000c
> Event: time 1566720565.547205, type 1 (Key), code 315 (BtnStart), value 0
> Event: time 1566720565.547205, -------------- Report Sync ------------
> Event: time 1566720566.731301, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720566.731301, type 1 (Key), code 314 (BtnSelect), value 1
> Event: time 1566720566.731301, -------------- Report Sync ------------
> Event: time 1566720566.735247, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720566.735247, type 1 (Key), code 314 (BtnSelect), value 0
> Event: time 1566720566.735247, -------------- Report Sync ------------
> Event: time 1566720566.923356, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720566.923356, -------------- Report Sync ------------
> Event: time 1566720567.915241, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1566720567.915241, type 1 (Key), code 304 (BtnA), value 1
> Event: time 1566720567.915241, -------------- Report Sync ------------
> Event: time 1566720568.075272, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1566720568.075272, type 1 (Key), code 304 (BtnA), value 0
> Event: time 1566720568.075272, -------------- Report Sync ------------
> Event: time 1566720569.779277, type 4 (Misc), code 4 (ScanCode), value 90002
> Event: time 1566720569.779277, type 1 (Key), code 305 (BtnB), value 1
> Event: time 1566720569.779277, -------------- Report Sync ------------
> Event: time 1566720569.923352, type 4 (Misc), code 4 (ScanCode), value 90002
> Event: time 1566720569.923352, type 1 (Key), code 305 (BtnB), value 0
> Event: time 1566720569.923352, -------------- Report Sync ------------
> Event: time 1566720570.419368, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720570.419368, type 1 (Key), code 314 (BtnSelect), value 1
> Event: time 1566720570.419368, -------------- Report Sync ------------
> Event: time 1566720570.423334, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720570.423334, type 1 (Key), code 314 (BtnSelect), value 0
> Event: time 1566720570.423334, -------------- Report Sync ------------
> Event: time 1566720570.563364, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720570.563364, -------------- Report Sync ------------
the map key:
> Event: time 1566720570.423334, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720570.423334, type 1 (Key), code 314 (BtnSelect), value 0
> Event: time 1566720570.423334, -------------- Report Sync ------------
> Event: time 1566720570.563364, type 4 (Misc), code 4 (ScanCode), value 9000b
> Event: time 1566720570.563364, -------------- Report Sync ------------
other key: like A
> Event: time 1566720567.915241, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1566720567.915241, type 1 (Key), code 304 (BtnA), value 1
> Event: time 1566720567.915241, -------------- Report Sync ------------
> Event: time 1566720568.075272, type 4 (Misc), code 4 (ScanCode), value 90001
> Event: time 1566720568.075272, type 1 (Key), code 304 (BtnA), value

is it a problem?

shantigilbert, 2019-08-25:
@betaxoi is the one that could awnser this

liangsdu, 2019-08-26:
i mean the map key report three times, **normal** key report two times @BetaXOi

liangsdu, 2019-08-30:
joystick can works on ES
on retroarch sometimes works, sometimes not
on fbaneo, the back key(remap to select, for insert coin) don't  work(most time, little time push two times work one time)

shantigilbert, 2019-09-02:
Sorry, I don't really know, this was a tool that was created by @BetaXOi and I've never used it :(

liangsdu, 2019-09-03:
forget this issue, i buy a new joystick

closed 2019-09-03 by liangsdu