EmuELEC/EmuELEC #43: build error. install libgpg-error has failed

bu2zhouzhu, 2019-09-21:
os: Manjaro
kernel: 4.19.69
#### Problem
I execute this command:
```
PROJECT=Amlogic-ng ARCH=arm DISTRO=EmuELEC make image
```
This build process successful run to [174/374] until 175: `BUILD libgpg-error`. Following is error:
```
gawk -f /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src/mkstrtable.awk -v textidx=2 -v nogettext=1 \
	-v prefix=GPG_ERR_ -v namespace=errnos_ \
	/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src/errnos.in >errnos-sym.h
/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/toolchain/bin/host-gcc -g -O0 -I. -I/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src -o mkheader /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src/mkheader.c
cat /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src/gpg-error.def.in >_gpg-error.def.h
gawk: fatal: cannot use gawk builtin `namespace' as variable name
make[3]: *** [Makefile:1617: errnos-sym.h] Error 2
make[3]: *** Waiting for unfinished jobs....
gawk: /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src/mkerrcodes.awk:88: warning: regexp escape sequence `\#' is not a known regexp operator
/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/toolchain/bin/armv8a-libreelec-linux-gnueabi-cpp -I. -I/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/src -I..   _gpg-error.def.h | \
  grep -v '^#' >gpg-error.def
rm _mkerrcodes.h
rm _gpg-error.def.h
make[3]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/.armv8a-libreelec-linux-gnueabi/src'
make[2]: *** [Makefile:509: all-recursive] Error 1
make[2]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/.armv8a-libreelec-linux-gnueabi'
make[1]: *** [Makefile:441: all] Error 2
make[1]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libgpg-error-1.36/.armv8a-libreelec-linux-gnueabi'
FAILURE: scripts/install libgpg-error has failed!
[175/374] [FAIL] install libgpg-error

The following logs for this failure are available:
  stdout: /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/.threads/logs/177/stdout
  stderr: /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/.threads/logs/177/stderr

parallel: This job failed:
package_worker 2 177 374 'install libgpg-error'
Parallel build failure - see log for details. Time of failure: Sat Sep 21 17:30:56 CST 2019
```
[177.zip](https://github.com/shantigilbert/EmuELEC/files/3638554/177.zip)

shantigilbert, 2019-09-21:
I do not use manjaro to build to its possible somethibg is not compatible, the log says the problem is with awk so maybe manjaro uses different awk?

bu2zhouzhu, 2019-09-22:
Find a similar problem on bugs.archlinux.org: [[gitlab] fails to build from source with gawk 5.x](https://bugs.archlinux.org/task/63654). I read the comments and don't know what they are talking about and how to fix it. Seems like it's a archlinux bug and can be fixed in the next version.

Guess my way to solve this problem is to install Ubuntu dual boot with Manjaro and build EmuELEC on Ubuntu.

bu2zhouzhu, 2019-09-22:
@shantigilbert You need add `sudo apt install texinfo` in your [README.md](https://github.com/shantigilbert/EmuELEC/blob/EmuELEC-CE.9.2/README.md), or build on Ubuntu 18.04 will fail because of `makeinfo command not found`.

```
  MAKEINFO /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/doc/libtool.info
/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/build-aux/missing: line 81: makeinfo: command not found
WARNING: 'makeinfo' is missing on your system.
         You should only need it if you modified a '.texi' file, or
         any other file indirectly affecting the aspect of the manual.
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
make[3]: *** [Makefile:1352: /home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/doc/libtool.info] Error 127
make[3]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/.x86_64-linux-gnu'
make[2]: *** [Makefile:1558: all-recursive] Error 1
make[2]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/.x86_64-linux-gnu'
make[1]: *** [Makefile:988: all] Error 2
make[1]: Leaving directory '/home/me/code/EmuELEC/build.EmuELEC-Amlogic-ng.arm-2.7/libtool-2.4.6/.x86_64-linux-gnu'
FAILURE: scripts/build libtool:host has failed!
[013/374] [FAIL] build   libtool:host
```

shantigilbert, 2019-09-23:
Closing this as it seems to be resolved

commit 93a1a071 (2019-10-31): Refactor some settings to be used in emuelec.conf

* Put some functions in profile.d 
* Change some settings to use emuele.conf
* This also fixes #43 by forcing video mode if EE_VIDEO_MODE file exists and its not empty (eg it includes the videomode inside)
* Remove some unused scripts 
* Rename package emulationstation-addon used only for the addon
closed 2019-09-23 by shantigilbert