EmuELEC/EmuELEC #46: Help Testing/Configuring the following emulator cores

shantigilbert, 2019-10-21:
There are a few platforms that have not been properly tested on EmuELEC, they are included but I either didn't knew how to test them or I don't have enough time to do so, could anyone help me by testing any of these platforms? or if you have tested any, can you share how you did it? was ist just placing the roms, or does it needs a BIOS etc.

The untested platforms are: 

```
AMSTRADCPC
ATARI800
C16/Plus4
C128
CVIC20
CANONBALL
CAVESTORY
CHAILOVE
DOOM
DAPHNE
FLASHBACK
GME
INTELLIVISION
NEC PC 9800
PCFX
QUAKE
RICKDANGEROUS
UZEBOX    
X68000
ZX81
```

asakous, 2019-10-22:
x68000 woks . put bios into /storage/roms/bios/keropi
NEC PC 9800 works . put bios and font file into /storage/roms/bios/np2kai

shantigilbert, 2019-10-22:
> x68000 woks . put bios into /storage/roms/bios/keropi
> 
> this core is not included by default .
> NEC PC 9800 works . put bios and font file into /storage/roms/bios/np2kai

Thanks, which core is that ?

asakous, 2019-10-24:
doom works
put doom.wad and  prboom.wad at same directory
quake works
put pak0.pak and pak1.pak into quake/id1 directory
canonball works
extract OutRun Revision B ROMs in the CannonBall directory and create a empty file call  CannonBall.game then select the game file to load game.

shantigilbert, 2019-10-24:
@asakous Thanks for the help! :)

edneyhelene, 2019-11-05:
I have tested the C16 and it works flawlessly \o

 ZX81 works too !

shantigilbert, 2019-11-05:
Thanks for the help @edneyhelene

edneyhelene, 2019-11-05:
> Thanks for the help @edneyhelene

Welcome \o

shantigilbert, 2019-11-07:
@edneyhelene did you had to do anything else other than placing the ROMS in the folder? bios? settings? etc

edneyhelene, 2019-11-07:
> @edneyhelene did you had to do anything else other than placing the ROMS in the folder? bios? settings? etc

hum not that i remember just inserted the files there

Kegg1701, 2019-11-12:
on installing Drastic on S905x (arm 3.0 generic) and rebooting, EmulationStation goes into a boot loop. 

On testing Drastic using coreelec & libreelec (modified to use my own personal retro gaming addon), Drastic starts perfectly, but on starting a game I get vertical black lines over the screen. Have been unable to test with emuelec properly due to ES boot looping after installing.. sorry

PPSSPP works, but suggest using a really old version for better performance from games for s905x etc.. I build against 1.5.4

Reicast works, but the sound is awful on most games, forever glitching out.. otherwise the usual graphical glitches you'd expect with Mali and the s905 box.

If you wish I can do a video for comparison and issues.

edneyhelene, 2019-11-12:
Which ppsspp build you think is better?

edneyhelene, 2019-11-12:
i dont have this bootloop problem on my s905X. The reicastSA gives me audio problems too, thats why i do use flycast, also the performance for me with that core is awful.
But other than that its working fantastically

Kegg1701, 2019-11-12:
1.5.4, for me anyway, really performs the best on these boxes

Dreamcast games do play pretty awful using Flycast and, again, I use a slightly older build of Reicast for doing so.. Naomi/Atomis with Flycast though is just amazing and works beautifully for many games I've tested.

Im going to test the NG build at some point.. got an S905x2 sat around here somewhere

edneyhelene, 2019-11-12:
Thats crazy!

For me flycast works wonderfully !

reicast gives me a super slow emulation lol

I'm on s905x too !

Kegg1701, 2019-11-12:
lol.. just tested Dreamcast on Flycast... forgot I updated the core for my build the other day... yeah.. agreed runs a lot better than the last time i tested it. Also disabling DSP makes the performance better.

edneyhelene, 2019-11-12:
Exactly =)

shantigilbert, 2019-11-12:
@Kegg1701 Thanks for the tips, but could you open a different issue for this? or better yet, join the discord and discuss it there? it has nothing to do with OP.

Kegg1701, 2019-11-12:
will do

shantigilbert, 2019-11-12:
@Kegg1701 was waiting for you to open the other issue, but I will reply here.

While PPSSPP 1.5.4 might be good for some games, in general it is slower that the newest PPSSPP on the S905 I have with 2gb of RAM, take for example Outrun 2006, on 1.5.4 it runs at the most 20fps (with some areas getting close to 30 for a few seconds) even at 1x resolution on 720p. Now on the newest PPSSPP it runs at a solid 30fps even at 2x resolution, at 1080p

I haven't tested many games on 1.5.4 but the ones I tested ran either the same or worse than on the newer PPSSPP, Unless I am missing something very obvious? (like a setting or something) I think you might need to revisit PPSSPP as you did with Flycast. 

Heck it it even runs God Of War at a decent playable frame rate following [this small guide](https://github.com/EmuELEC/EmuELEC/wiki/How-to-play-God-of-War-on-S905-with-standalone-PPSSPP-at-a-decent-speed%3F) by @SumavisionQ5 granted it is not full speed, but still very playable. 

Note that this is running EmuELEC stand alone I have not made any sort of testing with LE since 2 years or so ago. But it would be really good if you uploaded that video with the comparison, as I am really always open for any improvements!

Now for Drastic, not sure what is going on your side, but it installed and played correctly here. 
But Drastic needs to have a patched Kernel to work, it won't work on either LE nor CE (as add-on) since they use a Kernel with a modified alpha channel. (see https://github.com/EmuELEC/EmuELEC/blob/EmuELEC-3.0/packages/linux/patches/amlogic-3.14/linux-010_sx05re_set_colordepth.patch) if you find a way to make it work on either of them I would really like to know, but for now that is how I made it work on EmuELEC.

forest-b-sonic, 2019-12-25:
PCSX will flash back after 4-5 hours of operation.

howels, 2020-03-07:
Please update Ppsspp, the latest 1.9.3 release is much faster!

ducdreamer, 2020-03-14:
Hello there,
I have a problem running the MSX emulator.
it is not in the list but it is not running on my system
emuelec 3.4
T95Z plus
s912 2 gb ram gb ethernet wireless b/g/n
emuelec is running 1080P ps3 bluetooth controller
I have installed al the nessesary bios files.
After checking there are no errors there.
but when I try to run a MSX, MSX2 or MSX2 PLUS none of them work
after trying to run any rom you are back at the emuelec.
 the other systems I have tested are nes snes master system ps1
the are running ok
I am hoping for some tips.
MSX is my favorite system I used when I was young

shantigilbert, 2020-03-15:
Only reply if its RELATED TO THE ISSUE, if not , open another issue

BaltasarParreira, 2020-04-08:
DAPHNE/hypseus runs out of the box as long we put the right files (game roms) on the emu ROMs folder and create the <GAME>.daphne folders with the respective ripped video and audio files.
Also sound folder needs to have all necessary audio <xyz>.wav files, it's like an bios required files.
The only problem I'm facing and can't still figure out is the joy-pad mapping as I can't control anything in the games as by default works with keyboard.

BaltasarParreira, 2020-04-08:
Ok I figure out the joypad mapping, at least for me tha tI have a BT PS3 controller.
Editing the file:
\\<your IP>\Configfiles\emuelec\configs\hypseus\hypinput.ini

On the file the 3rd column is the joypad as It's mentioned on the header inside the file.
Here you have to put the corresponding number for the controller pressed button. You can try to find these numbers by SSH to your Emuelec and run this command:
jstest --event /dev/input/js0

My settings fille became like this:
```
# Sample hypinput.ini
# All key options listed
# Arcade layout by OzFalcon
# Hypseus uses SDL2 Keycodes
# Button3 Turns scoreboard on/off
# SWITCH = Key1 Key2 Joy

[KEYBOARD]
KEY_UP = 1073741906 0 5 # pad button UP
KEY_DOWN = 1073741905 0 7 # pad button DOWN
KEY_LEFT = 1073741904 0 8 # pad button LEFT
KEY_RIGHT = 1073741903 0 6 # pad button RIGHT
KEY_COIN1 = 53 54 1 # SELECT button
KEY_COIN2 = 54 53 0
KEY_START1 = 49 0 4 #START button
KEY_START2 = 50 0 0
KEY_BUTTON1 = 1073742048 0 15 # X button
KEY_BUTTON2 = 1073742050 0 14 # CIRCLE button
KEY_BUTTON3 = 32 0 13 # TRIANGLE button
KEY_SKILL1 = 1073742049 0 0
KEY_SKILL2 = 122 0 0
KEY_SKILL3 = 120 0 0
KEY_SERVICE = 57 0 0 
KEY_TEST = 1073741883 0 0
KEY_RESET = 48 0 0
KEY_SCREENSHOT = 1073741893 0 0
KEY_QUIT = 27 0 17 # PS HOME button
KEY_PAUSE = 112 0 0
KEY_CONSOLE = 92 0 0
KEY_TILT = 116 0 0
END
```
NOTE:  Zero (0) is disabled so since there is most of joypads a zero button we here on this conf 0 =1, 1 = 2 and so on, resuming you have to add 1 to the value you get from the jstest command.

Kegg1701, 2020-04-08:
nice one.. i never got round to see if the ini would actually affect controller mapping.. literally have not touch my OGA in weeks... least now i won't have to bother with creating an OGA specific version of hypseus,which there is, just not done a PR as things are pretty busy here for quite some time to come. cheers

BaltasarParreira, 2020-04-08:
UZEBOX, for now loading a rom file directly from retroarch using the uzem core, works.
Going to test further to see how to automate this from emulationstation.

BaltasarParreira, 2020-04-09:
**How to add UZEBOX to emulationstation.**

_1- edit the folowing file:_
`\\<your IP>\Configfiles\emulationstation\es_systems.cfg`
_2- add before the last line (`</systemList>`) this:_
```
    <system>
        <name>uzebox</name>
        <fullname>uzebox</fullname>
        <path>/storage/roms/uzebox</path>
        <extension>.uze .UZE .hex .HEX</extension>
        <command>/emuelec/scripts/emuelecRunEmu.sh %ROM% -P%SYSTEM% --controllers="%CONTROLLERSCONFIG%"</command>    
        <platform>uzebox</platform>
        <theme>uzebox</theme>
    </system>
```
_3. save it_
_4- edit this file:_
`\\<your IP>\Configfiles\emulationstation\scripts\getcores.sh`
_5- add this:_
```
"uzebox")
	CORES="uzem_libretro"
	;;
```
_before this lines at the end:_
```
esac

[[ $2 == "default" ]] && CORES="$(echo ${CORES} | cut -d',' -f1)"

echo $CORES
```
_6- save it_
_7- create the uzebox rom folder in:_
`\\<your IP>\roms`
like this:
`\\<your IP>\roms\uzebox`
_8- put the game rom files inside the previous created folder_
_9- restart emulationstation_
_10- new game group will show without any graphics as the theme's don't support it yes, bu you can browse your games and lunch them as normal._

That's it !!
Someone care to try it and see if I miss anything?
At least It's working like a charm for me. 
; )

shantigilbert, 2020-06-28:
Guess the rest of the cores are not important

closed 2020-06-28 by shantigilbert