EmuELEC/EmuELEC #48: Kernel

Power535, 2019-10-28:
How to find out ?
PKG_VERSION="xxxxx"
Link for kernel ?
PKG_URL="https://github.com/CoreELEC/.../archive/$PKG_VERSION.tar.gz"
I want try replace kernel
https://github.com/150balbes/Amlogic_s905-kernel/tree/4.20

shantigilbert, 2019-10-28:
Git hash

Power535, 2019-10-28:
I do not understand
Please examples

shantigilbert, 2019-10-28:
If you do not understand that, I don't think you should be changing the Kernel, as you are going to have to modify a lot of other stuff. 

Can I ask what is the purpose of changing the kernel?

Power535, 2019-10-28:
I know as compile kernel
My old monitor not support

Please examples PKG_VERSION and PKG_URL for
https://github.com/150balbes/Amlogic_s905-kernel/tree/4.20

shantigilbert, 2019-10-28:
PKG_VERSION="bca47b8cea5a6401282353deb1baffd2f52078a6"
PKG_URL="https://github.com/150balbes/Amlogic_s905-kernel/archive/$PKG_VERSION.tar.gz"

Let me know if you are successful

Power535, 2019-10-28:
How are these links obtained ?
Please examples

closed 2019-10-28 by shantigilbert