EmuELEC/EmuELEC #49: CVBS not working on X96 s905W

edneyhelene, 2019-10-31:
Hello, i cant make it work.

it boots up, then i can see the emuelec logo on the old tv, and then black screen.
I tried to add EE_VIDEO_MODE file to the sdcard on the same folder as dtb.img but nothing changes. 

I really need to make it work on this old tv, its a must for me, i'm desperate.

The box is the X96 S905X with 2gb ram and 16gb rom.

Please help?

edneyhelene, 2019-10-31:
Solution found!
Steps to fix the problem:

Open nautilus as root on linux.
Navigate to STORAGE/.config/emulationstation/ es_settings.cfg
and replace the resolution to match your tv resolution:

in my case it was <string name="EmuELEC_VIDEO_MODE" value="480cvbs" /> 
inserted 480cvbs and it works!

Problem closed

closed 2019-10-31 by edneyhelene

shantigilbert, 2019-10-31:
This is now fixed on latest commits, EE_VIDEO_MODE takes priority

varma86, 2020-03-13:
Set 2160p60hz and it worked. Might be a incompatibility with my Hitachi TV.
Very strange because CoreElec works pretty well in 1080p too. Android too is set to 1080p.
I hope you will investigate. Maybe I could help with some log files, if you instruct me how to.

Sidordroid, 2020-11-03:
EE_VIDEO_MODE and what folder should this file be put in? And what should be written in it?I guess 480cvbs 576cvbs?