EmuELEC/EmuELEC #50: ROM folders should be identical with Retroarch Database

asakous, 2019-11-04:
for people who use Lakka 
they can just use Retroarch RDB files to add playlists
and thumbnails for image preview

shantigilbert, 2019-11-04:
ROMS folders are identical to Retropie, can you elaborate on which ROM folders are different?

asakous, 2019-11-04:
These are folder name that use in Retroarch Databse 
Amstrad - CPC
Atari - 2600
Atari - 5200
Atari - 7800
Atari - Jaguar
Atari - Lynx
Atari - ST
Bandai - WonderSwan Color
Bandai - WonderSwan
Cannonball
Casio - Loopy
Casio - PV-1000
Cave Story
ChaiLove
Coleco - ColecoVision
Commodore - 64
Commodore - Amiga
Commodore - Plus-4
Commodore - VIC-20
Dinothawr
DOOM
DOS
Emerson - Arcadia 2001
Entex - Adventure Vision
Epoch - Super Cassette Vision
Fairchild - Channel F
FBNeo - Arcade Games
Flashback
Funtech - Super Acan
GamePark - GP32
GCE - Vectrex
Handheld Electronic Game
Hartung - Game Master
LeapFrog - Leapster Learning Game System
Lutro
Magnavox - Odyssey2
MAME 2000
MAME 2003-Plus
MAME 2003
MAME 2010
MAME 2015
MAME 2016
MAME
Mattel - Intellivision
Microsoft - MSX
Microsoft - MSX2
Microsoft - Xbox
MrBoom
NEC - PC Engine - TurboGrafx 16
NEC - PC Engine CD - TurboGrafx-CD
NEC - PC Engine SuperGrafx
NEC - PC-FX
Nintendo - e-Reader
Nintendo - Family Computer Disk System
Nintendo - Game Boy Advance
Nintendo - Game Boy Color
Nintendo - Game Boy
Nintendo - GameCube
Nintendo - Nintendo 3DS (Digital)
Nintendo - Nintendo 3DS
Nintendo - Nintendo 64
Nintendo - Nintendo 64DD
Nintendo - Nintendo DS
Nintendo - Nintendo Entertainment System
Nintendo - Pokemon Mini
Nintendo - Satellaview
Nintendo - Sufami Turbo
Nintendo - Super Nintendo Entertainment System
Philips - Videopac+
Quake
RCA - Studio II
Rick Dangerous
RPG Maker
ScummVM
Sega - 32X
Sega - Dreamcast
Sega - Game Gear
Sega - Master System - Mark III
Sega - Mega Drive - Genesis
Sega - Mega-CD - Sega CD
Sega - PICO
Sega - Saturn
Sega - SG-1000
Sharp - X68000
Sinclair - ZX 81
Sinclair - ZX Spectrum +3
Sinclair - ZX Spectrum
SNK - Neo Geo CD
SNK - Neo Geo Pocket Color
SNK - Neo Geo Pocket
Sony - PlayStation
The 3DO Company - 3DO
Thomson - MOTO
TIC-80
Tiger - Game.com
Tomb Raider
Uzebox
VTech - CreatiVision
VTech - V.Smile
Watara - Supervision
 
When folder name is identical  to Retroarch. so that Lakka Scan to List function will work.
When rom set come from Nointro.   thumbnails  and cheats will work too.

shantigilbert, 2019-11-04:
I think you are confusing the rom paths, that database is based on the rom names not the path, they can be anywhere in your hard disk as long as they use the no intro name

shantigilbert, 2019-11-04:
I think I know what the problem is, I do not include the database files, you need to download them in Retroarch first

asakous, 2019-11-04:
unless the folder name is identical to the Retroarch  one. otherwise Retroarch thumbnails can not be used(or manually rename ).Please close issus. My primary GUI interface is xmb so I think follow Retroarch rules maybe is a good idea.

shantigilbert, 2019-11-04:
I think I am not understanding this, why would it need to be changed in EmuELEC? if you only use Retroarch you can have your ROMS in any folder, including the ones you mentioned. I've never heard of these folder requirements for Retroarch before, but on the other hand I haven't setup Retroarch to be my main fronted.

closed 2019-11-05 by asakous