EmuELEC/EmuELEC #59: About emuelec3.1, the question of the shortcut function key in retroarch mode?

ji125443, 2019-11-19:
Volume shortcut function key code: - 3 and + 3 are the default corresponding to the up key and down key of my handle, but I have changed to other keys, and the up key and down key volume adjustment will still work, which makes me quite puzzled when playing games. The previous version will not have such a problem. It is better to map all the function shortcut keys to the keyboard
[image]
[image]
[image]

shantigilbert, 2019-11-19:
I am not sure I follow, you mean the volume changes when you move your dpad? these buttons should only work when in combination of a Hotkey (hotkey+up = volume up, hotkey+down = volume down, etc) are you setting a hotkey on initial ES configuration?

ji125443, 2019-11-20:
I have already initialized ES, whether to turn on the shortcut hot key to select "no", but when playing the game, press the up key volume alone will still become larger, should not be a shortcut hot key problem

shantigilbert, 2019-11-20:
You need to set a hotkey

ji125443, 2019-11-20:
ok

shantigilbert, 2019-11-20:
Will try to find a workaround for people that do not use a hotkey for whatever reason

shantigilbert, 2019-11-20:
Please replace this file 
`/storage/.emulationstation/scripts/configscripts/retroarch.sh` 

with this one

https://github.com/EmuELEC/EmuELEC/blob/5b2de52c149f5bfc40e1e974fa53174cbbd2976d/packages/sx05re/emuelec-emulationstation/config/scripts/configscripts/retroarch.sh

then reconfigure your gamepad in ES and try it out, let me know if it fixes the problem.

ji125443, 2019-11-20:
ok，Has been resolved, thank you

closed 2019-11-20 by shantigilbert