EmuELEC/EmuELEC #60: [Bug]Loading game freezed after quit from RetroArch

MiltonLai, 2019-11-20:
**Version**
3.1 (also in 3.0)

**How to reproduce:**
Run a clean 3.1 system, fill it with some game roms
in ES, Quit -> Start RetroArch
In RetroArch, make some modifications (I changed the video aspect ratio to 16:9). Save Current Configuration, Quit RetroArch.
When back in ES, try to run any roms ( I tried the roms under arcade and nes), the screen will freeze immediately, no response to any keys except Ctrl+Alt+Del

shantigilbert, 2019-11-20:
Can you supply aby logs? Emuelec.log would be helpful to determine this

MiltonLai, 2019-11-21:
Thanks for your reply! I attached the emuelec.log and retroarch.cfg
[emuelec_log&retroarch_cfg.zip](https://github.com/EmuELEC/EmuELEC/files/3872745/emuelec_log.retroarch_cfg.zip)

shantigilbert, 2019-11-23:
@MiltonLai I cannot reproduce this error, went into Retroarch, made a few changes (including the aspect ratio) saved the Retroarch.cfg and it still works as it should. Your the logs look normal

MiltonLai, 2019-11-24:
Thank you for looking into it. I am not sure if it is a hardware related issue, currently I only tested it on a R3300-L tv box, maybe I will find other devices for testing in next week. 
I have done more tests today and got some new findings. The freeze would 100% happen if quit -> start retroarch -> quit retroarch -> run a rom, or try to restart es. I guess there is something blocking the new process from running, because when I press ctrl+alt+del, the screen will show the loading picture for a very very short time before it totally black out. And I found a walk round: after quit from retroarch, change the ui language, which will restart es, then it is safe to run the roms. 
I also checked the log, but there is merely any valuable output. Can it be more verbose by raising the log level and where to change it?

shantigilbert, 2019-11-24:
I am not familiar with that device, what chip (SOC) does it have?

It sounds to me like it's a framebuffer issue like the s922x had.

MiltonLai, 2019-11-25:
The chip is S905L, with 1g ram + 8g rom and RTL8189. 
I made a clean install on a 32g TF card and tried again. dino.zip is the rom I opened to reproduce the freeze. I found this log after reboot from freeze, but not quite sure if it is related:
`EmuELEC:~ # tail -f -n 100 /emuelec/logs/emuelec.log 
EmuELEC Run Log
PLATFORM: arcade
ROM NAME: /storage/roms/arcade/dino.zip
BASE ROM NAME: dino.zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/arcade/dino.zip
2nd Argument: -Parcade
3rd Argument: 
4th Argument: 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
Emulator Output is:
[WARN] --libretro argument "/tmp/cores/fbneo_libretro.so" is neither a file nor directory. Ignoring.
[INFO] RetroArch 1.8.1 (Git f2898f4404)
[INFO] Redirecting save file to "/storage/roms/arcade/dino.srm".
[INFO] Redirecting savestate to "/storage/roms/arcade/dino.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Nov 19 2019
[INFO] Version: 1.8.1
[INFO] Git: f2898f4404
[INFO] =================================================
[ERROR] Frontend is built for dynamic libretro cores, but path is not set. Cannot continue.
[ERROR] Fatal error received in: "init_libretro_symbols()"
`
`EmuELEC:~ # ls -l /tmp/cores/fbneo_libretro.so
-rwxr-xr-x    1 root     root      36786244 Nov 19 00:41 /tmp/cores/fbneo_libretro.so`
The file does exist, I don't know why it reports this warning.

shantigilbert, 2019-11-25:
can you run from SSH this

```
systemctl stop emustation
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
```
and see what it says?

MiltonLai, 2019-11-25:
The output
```
EmuELEC:~ # systemctl stop emustation
EmuELEC:~ # /usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
[INFO] RetroArch 1.8.1 (Git f2898f4404)
[INFO] Redirecting save file to "/storage/roms/arcade/dino.srm".
[INFO] Redirecting savestate to "/storage/roms/arcade/dino.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Nov 19 2019
[INFO] Version: 1.8.1
[INFO] Git: f2898f4404
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/fbneo_libretro.so"
[INFO] [Overrides] no core-specific overrides found at /storage/.config/retroarch/config/FinalBurn Neo/FinalBurn Neo.cfg.
[INFO] [Overrides] no content-dir-specific overrides found at /storage/.config/retroarch/config/FinalBurn Neo/arcade.cfg.
[INFO] [Overrides] no game-specific overrides found at /storage/.config/retroarch/config/FinalBurn Neo/dino.cfg.
...
```
The game starts normally.

shantigilbert, 2019-11-25:
can you start emulationstation again and see if you can run it from there after that? 

`systemctl start emustation`

MiltonLai, 2019-11-25:
I run the following commands repeatedly, didn't find any errors
```
systemctl stop emustation
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
(double click quit key to quit)
systemctl start emustation
systemctl stop emustation
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
(double click quit key to quit)
systemctl start emustation
```
If I run the following commands, the game image will be overlapped by ES UI
```
systemctl start emustation
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
(double click quit key to quit)
/usr/bin/retroarch -v -L /tmp/cores/fbneo_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/arcade/dino.zip
```

shantigilbert, 2019-11-25:
yeah I meant start the game from ES after restarting it.

I tried several times, with several games and I just cannot reproduce the issue, I am going to have to assume its something on your side, probably bad SD card? 

what is the reason you want to modify the settings on RA ? you can set the aspect ratio inside ES and it applies to all games in retroarch, or just the system or game

MiltonLai, 2019-11-25:
No, it seems nothing to do with changing RA settings, even if I just start RA and quit, then running a rom will freeze. I guess the problem resides in ES, because when it freezes, I can still see the /usr/bin/emulationstation process in ps command, and if I kill this process, the ES will restart and everything goes find after that.
Thank you for looking into it.
Maybe it is a device related problem, I will try it on other tv boxes later this week and keep you updated.

shantigilbert, 2019-11-25:
Ok, please do as I am interested in fixing this if its a general issue.

SumavisionQ5, 2019-11-26:
你试试修改/storage/.config/emuelec/scripts/emuelecRunEmu.sh看看。
把
‘
# Only run fbfix on N2
[[ ! -f "/ee_s905" ]] && /storage/.config/emuelec/bin/fbfix

# Show exit splash
${TBASH} /emuelec/scripts/show_splash.sh exit

# Kill jslisten, we don't need to but just to make sure 
killall jslisten
’
这几行注释掉，看看还卡不卡。
PS：你的英语不错，可以上discord与shantigilbert交流，那样更方便。

shantigilbert, 2019-11-26:
Fbfix does not exist on s905

SumavisionQ5, 2019-11-26:
> Fbfix does not exist on s905

Sorry, I thought he had same issue as me. My issue is that it takes a long time to go back to ES after quiting the games. I have an S905L box, and I will try to reproduce his issue tonight.

shantigilbert, 2019-11-26:
Yeah but, fbfix does not exist at all on s905 devices, as it is not needed. I thought you had your issue fixed.

SumavisionQ5, 2019-11-26:
I doubt whethe it is an issue related with vlc or not.
I used to use mpv and everything was all right then.
@MiltonLai  Can you try to comment out line with vlc in showspalsh.sh and see whether issue happens again?

MiltonLai, 2019-11-26:
@SumavisionQ5 I hadn't use discord for a long time... maybe I will install and try later. I tried your instructions, 
* Comment out 2 lines as below. It doesn't work. 
```
if [[ -f "/storage/.config/emuelec/configs/novideo" ]] && [[ ${VIDEO} != "1" ]];
        if [ "$PLATFORM" != "intro" ]; then                                     
        (                                                                       
#                       ply-image $SPLASH > /dev/null 2>&1                      
        )&                                                                      
        fi                                                
else                                                      
        SPLASH="/usr/config/splash/emuelec_intro_1080p.mp4"                     
        set_audio alsa                                                          
        [ -e /storage/.config/asound.conf ] && mv /storage/.config/asound.conf /
        vlc -I "dummy" --aout=alsa $SPLASH vlc://quit < /dev/tty1               
        touch "/storage/.config/emuelec/configs/novideo"                        
# VLC cleans up the FB after exiting, which means we need to show a loading imag
        SPLASH="/storage/.config/splash/splash-1080.png"                        
#       ply-image $SPLASH > /dev/null 2>&1                                      
[ -e /storage/.config/asound.confs ] && mv /storage/.config/asound.confs /storag
fi
```
* Comment out these lines, it doesn't work
```
# Only run fbfix on N2                                                    
#[[ ! -f "/ee_s905" ]] && /storage/.config/emuelec/bin/fbfix
                                                                                
# Show exit splash                                                              
#${TBASH} /emuelec/scripts/show_splash.sh exit                                  
                                                                                
# Kill jslisten, we don't need to but just to make sure
#killall jslisten             
```
* Disable splash in ES menu --doesn't work either

MiltonLai, 2019-11-26:
So far I tested it on other two devices Q5(S905M) and N1(S905D, boot with modified kernel.img & autoscripts), both have this issue.

SumavisionQ5, 2019-11-26:
Thanks for testing. We may need to investigate more on it.

shantigilbert, 2019-11-26:
What do you mean modified kernel?

MiltonLai, 2019-11-26:
> 
> 
> What do you mean modified kernel?

The default files in EMUELEC partition can not boot N1 box by applying the dtb files from device_tree folder (the closest one might be gxl_p230_2g.dtb I guess). So someone replaces the files with those from early version and that works. 
I just borrow their tricks because I have no idea of doing with it.

The files list, they replace everything except SYSTEM and SYSTEM.md5
[image]

N1 hardware specs:
CPU: Amlogic S905, ARM Cortex-A53 quad core up to 2.0GHz
GPU: ARM Mali™-450, 5core
ETH: RTL8211F 100/1000M
WLAN: Broadcom CYW43455 2.4GHz + 5GHz + BT4.1
ROM: Samsung KLM8G1GEME or FORESEE NCEMAM6G-08G，eMMC5.1 MLC = 8GB
RAM: Samsung K4B4G1646E-BYMA, 4Gb(512MB) DDR3-1866MHz * 4 = 2GB

shantigilbert, 2019-11-26:
that is interesting... do you know where can I download those files? the issue might be related to that

MiltonLai, 2019-11-26:
The files are attached. These files might not be related because the issue also exists in original clean installation.
[EMUELEC_replacement.zip](https://github.com/EmuELEC/EmuELEC/files/3890330/EMUELEC_replacement.zip)

SumavisionQ5, 2019-11-26:
@MiltonLai I can not reproduce your issue. I test offical 3.1 on Sumavision Q5 (both S905L and S905M) and BestTV R-3300L(S905L). They all works properly. Maybe it's your TF's issue? Did you try any other TF or SD card?

shantigilbert, 2019-11-26:
Also, maybe try booting from USB

MiltonLai, 2019-11-27:
I tried another TF card and did the following steps to reproduce this issue:
Write EmuELEC-Amlogic.arm-3.1-Generic.img to a 16g TF card with Disk Image Writer in Ubuntu, 
Copy gxl_p212_1g_slowemmc.dtb from device_trees and replace the img.dtb file in root folder,
Boot the box R3300-L with this TF card, wait it to resize and restart,
Setup the keyboard,
Connect the box to wired lan network, got the IP from system menu,
Upload some nes roms from Ubuntu,
Run a rom, everything looks okay, quit to ES,
In ES, Quit->Start RA,
In RA, quit,
Run a rom -- freeze.

@SumavisionQ5 Could you please create a img of your working tf card for R3300-L, with only a few nes roms in it, compress and upload it to pan.baidu.com so that I can restore and test it?

MiltonLai, 2019-11-27:
> Also, maybe try booting from USB

I tried, still got the same issue.

SumavisionQ5, 2019-11-27:
> I tried another TF card and did the following steps to reproduce this issue:
> Write EmuELEC-Amlogic.arm-3.1-Generic.img to a 16g TF card with Disk Image Writer in Ubuntu,
> Copy [...]

You said you used a keyboard? Do you have a joypad such as generic USB gamepad?
I did what you wrote, but everything is all right.
Very weird.
By the way, can you try to comment out all JSLISTEN related lines in EmuelecRunEmu.sh? I am afraid jslisten may break due to lacking of a joypad.

MiltonLai, 2019-11-27:
No luck, I commented out 3 lines that contains "jslisten", then the game won't start and the issue still exists. I will find a joystick and try it later.

MiltonLai, 2019-11-27:
I didn't find one js in office, so I tested it in previous releases
* 2.5.4: OK
* 2.7.1: OK
* 3.0: same issue
Maybe it is caused by some new feature introduced in 3.0?

shantigilbert, 2019-11-27:
There is definitely something different between our devices and yours, I cannot reproduce this, on none of my 6 devices, no matter what I do, and nobody else seems to have this problem. I am not sure what else to do.

shantigilbert, 2019-11-28:
Do you only have a keyboard? no joystick at all? you configure your keyboard in ES before starting Retroarch?

asakous, 2019-11-28:
not sure. but worth to try
delete cache_directory in retroarch.conf or leave it empty.

MiltonLai, 2019-11-28:
> Do you only have a keyboard? no joystick at all? you configure your keyboard in ES before starting Retroarch?

Yes, that is what I always did.... a joystick in office will catch too much attention, and I prefer playing games with keyboard (except those combat games like kof serials). I have a wireless joystick at home, I will test with them later.

shantigilbert, 2019-11-28:
Lol :P I will che k with just a keyboard that might be the issue

SumavisionQ5, 2019-11-28:
> > Do you only have a keyboard? no joystick at all? you configure your keyboard in ES before starting Retroarch?
> 
> Yes, that is what I always did.... a joystick in office will catch too much attention, and I prefer playing games with keyboard (except those combat games like kof serials). I have a wireless joystick at home, I will test with them later.
> 
> > not sure. but worth to try
> > delete cache_directory in retroarch.conf or leave it empty.
> > note: stop retroarch service before edit it.
> > The cache directory in retroarch.cfg is '/tmp/cache', this directory is empty
> 
> ```
> EmuELEC:~/.config/retroarch # cd /tmp/cache/
> EmuELEC:/tmp/cache # ll
> total 0
> EmuELEC:/tmp/cache # ll -a
> total 0
> drwxr-xr-x    2 root     root            40 Apr 11  2019 .
> drwxrwxrwt   16 root     root           340 Nov 27 21:37 ..
> ```

You misunderstood what asakous said. He said you could edit storage./config/retroarch/retroarch.cfg, delete cache_directory line and try again.

shantigilbert, 2019-11-28:
I thought maybe it was keyboard related, but no, I configured a fresh new 3.1 with just the keyboard and I could not reproduce this.

MiltonLai, 2019-11-29:
@SumavisionQ5 Sorry I didn't notice that. I just tested that but the issue still exists.

@shantigilbert I tried with joystick attached only but the issue still exists, so keyboard is not the cause.
A little progress is that I possibly found some clue, but I don't know why: 
* After quit from RA, if I stop the pulseaudio service (run 'systemctl stop pulseaudio'), then the rom can be started normally. 
* After quit from RA, run a rom, freeze. Then if I stop pulseaudio service, the rom will be started normally.

I noticed that when RA is running a rom, pulseaudio is not in the process list, so the issue may be caused by:
* Pulseaudio service prevents RA from starting a rom
* Pulseaudio service has trouble quit itself

Which script I should look into if I want to track down it?

SumavisionQ5, 2019-11-29:
> @SumavisionQ5 Sorry I didn't notice that. I just tested that but the issue still exists.
> 
> @shantigilbert I tried with joystick attached only but the issue still exists, so keyboard is not the cause.
> A little progress is that I possibly found some clue, but I don't know why:
> 
> * After quit from RA, if I stop the pulseaudio service (run 'systemctl stop pulseaudio'), then the rom can be started normally.
> * After quit from RA, run a rom, freeze. Then if I stop pulseaudio service, the rom will be started normally.
> 
> I noticed that when RA is running a rom, pulseaudio is not in the process list, so the issue may be caused by:
> 
> * Pulseaudio service prevents RA from starting a rom
> * Pulseaudio service has trouble quit itself
> 
> Which script/command will be executed before emuelecRunEmu.sh when clicking on a rom? I can confirm that the freeze happens before executing emuelecRunEmu.sh.

Just comment out line 13 to line 19, and line 254 in emuelecRunEmu.sh and try again. If Pulseaudio causes it, your issue will be gone.

shantigilbert, 2019-11-29:
This could be the issue, but its strange that it doesn't happen to anyone else :( 

I am pushing a test update, with a new script called "start_retroarch.sh" which will change audio to alsa before starting retroarch, but I assume by your description this probably solves nothing. 

The way this works is like this "emustation-config" changes the audio to pulse so that sound mixing works in ES, this is done before ES starts. Every time a game is run, emuelecRunEmu.sh changes the audio to ALSA before it starts, except in some cases where pulse is needed (like Scummvm) and changes it back to pulse when the game is over. 

So if your theory is correct, something is preventing pulse from either being killed or switching to ALSA.

As @SumavisionQ5 said, either comment those lines, or put 

```
systemctl stop pulseaudio
export SDL_AUDIODRIVER=alsa
```
 somewhere at the very top of emuelecRunEmu.sh, if you theory is correct it should work, then we can figure out what is causing it in your device.

SumavisionQ5, 2019-11-29:
Just a little off topic: I get a short buzz when quitting or starting a game on s905 box.  That may be caused by pulseaudio, and S922x is all right with it.

robertini1801, 2019-11-29:
Hi, I have the same problem.

shantigilbert, 2019-11-29:
@SumavisionQ5#0205 the buzz only happens on libretro? If so that is because I settled the volume to high, next test version should have this fixed. But it has nothing to do with pulse.

MiltonLai, 2019-11-29:
@SumavisionQ5 @shantigilbert I tried, but they don't work. The freeze happens before entering emuelecRunEmu.sh. I put a test line at the top of it like this, but test.log file didn't show up when screen freezes.
```
#!/bin/bash
echo 'test' > /emuelec/logs/test.log
# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 2019-present Shanti Gilbert (https://github.com/shantigilbert)

# Source predefined functions and variables
. /etc/profile
```

shantigilbert, 2019-11-29:
Before quiting  to retroarch, ssh and kill pulseaudio, then start retroarch and see if the same thing happens

MiltonLai, 2019-11-29:
It doesn't work, starting retroarch will also start pulseaudio:
* systemctl stop pulseaudio -> ssh: run 'ps aux' shows pulseaudio is gone
* quit->start retroarch -> ssh: run 'ps aux' shows pulseaudio is there
* quit retroarch -> ssh: run 'ps aux' shows pulseaudio is there
* run a rom -> freeze

shantigilbert, 2019-11-29:
Ok, try one more thing, edit the file emustation-config and change the set_audio to alsa

MiltonLai, 2019-11-29:
Cool, this works!

shantigilbert, 2019-11-29:
Now the problem is trying to figure out why it only happens to a few, I need to change this anyways but it's strange that it happens.

shantigilbert, 2019-11-29:
@MiltonLai thanks for helping figuring out the issue!

MiltonLai, 2019-11-29:
You are welcome. Thank you so much for your help!

closed 2019-11-29 by MiltonLai

mvventura, 2020-01-07:
Please, can someone help me?? i can't boot emuelec

i have a tx9 with s905M-B 4gb ram processor

what would be the dtb file ???

commit 5d164139 (2019-11-29): Retroarch: Set audio to alsa when starting RA directly

This fixes #60