EmuELEC/EmuELEC #61: How to adjust overscan for CRT?

25minutes, 2019-11-29:
I'm using latest EmuELEC v3.1 standalone version on a S905M device. I've followed the instructions in Wiki to create EE_VIDEO_MODE file to enable CRT output, which is working. Now I'd like to adjust overscan to optimize the display. However, I cannot find the info in Wiki.

I also own a Raspberry Pi 3 with RetroPie installed. I can edit config.txt to play with overscan_left/overscan_right/overscan_top/overscan_buttom settings to meet my needs.

Is it possible to do similar overscan adjustment on EmuELEC? Any suggestions are truly appreciated.

shantigilbert, 2019-11-29:
This is something that I have not investigated, I am pretty sure there is a command to do it, but I unfortunately do not know it. 

I will try to search for a solution, but since I do not have a CRT I depend on other people to test it.

25minutes, 2019-11-29:
Thanks shantigilbert. I can test it, just drop me a note when you have anything. 

FYI. Below is the doc I used for Raspberry Pi (RestroPie). In addition to the parameters I mentioned earlier, it also requires overscan_scale=1, in order to make EmulationStation/RetroArch to obey the overscan settings. 
https://retropie.org.uk/docs/Overscan/

keeno1982, 2019-12-05:
I need it too. My TV have no point to point mode so can not work perfect on 1080P. Raspberry overscan setting is good for this.

shantigilbert, 2019-12-23:
Sorry I can't find any info on this, I tried compiling fbset (full as the one included from busybox is very limited) but I still was not able to correct the overscan. I am out of ideas.

25minutes, 2019-12-24:
Thank you very much for taking the time on investigating it. I truly appreciate that.

Do you prefer I close the ticket for now, or leave it open? (in case someone happens to know a solution, he/she can add input)

shantigilbert, 2019-12-24:
Leave it open, hopefully someone could help :)

SumavisionQ5, 2019-12-24:
The answer is alway there, in setres.sh! Read it carefully, and you will find an answer in 480cvbs or 576cvbs. I did not answer this issue becasue I hate someone might use EmuELEC for commerical purpose.

25minutes, 2019-12-24:
Thanks SumavisionQ5. I have 2 questions:

1. I find a similar discussion at https://forum.odroid.com/viewtopic.php?f=112&t=17774&p=118897#p118897.  The basic idea is to set /sys/class/graphics/fb0/window_axis & /sys/class/graphics/fb0/free_scale. It's for odroid c1, but the last post of the thread says it works on odroid c2 (which seems to be s905) as well. 

However, When I test it on EmuELEC (S905M), it doesn't work. It turns out that I also have to set free_scale_axis, scale_width, scale_height. Do you know why? 

2. I see both 480cvbs & 576cvbs use: fbset -fb /dev/fb0 -g 1280 960 1280 1920 $BPP. Does it have to be 1280x960? Can I use other values? Any tips for choosing these values? I'm asking because it's not really a common resolution.

shantigilbert, 2019-12-24:
did this fix your issue?

shantigilbert, 2019-12-24:
Yep that fixed the issue for me as well

@25minutes in my case I had to use this settings

```
	fbset -fb /dev/fb0 -g 1280 960 1280 1920 32
	echo 0 0 1279 959 > /sys/class/graphics/fb0/free_scale_axis
	echo 30 10 669 469 > /sys/class/graphics/fb0/window_axis
	echo 0x10001 > /sys/class/graphics/fb0/free_scale
```
so I am guessing the values depend on your monitor/tv/screen

SumavisionQ5, 2019-12-25:
> Thanks SumavisionQ5. I have 2 questions:
> 
> 1. I find a similar discussion at https://forum.odroid.com/viewtopic.php?f=112&t=17774&p=118897#p118897.  The basic idea is to set /sys/class/graphics/fb0/window_axis & /sys/class/graphics/fb0/free_scale. It's for odroid c1, but the last post of the thread says it works on odroid c2 (which seems to be s905) as well.
> 
> However, When I test it on EmuELEC (S905M), it doesn't work. It turns out that I also have to set free_scale_axis, scale_width, scale_height. Do you know why?
> 
> 1. I see both 480cvbs & 576cvbs use: fbset -fb /dev/fb0 -g 1280 960 1280 1920 $BPP. Does it have to be 1280x960? Can I use other values? Any tips for choosing these values? I'm asking because it's not really a common resolution.

https://www.ppxclub.com/home.php?mod=space&uid=150233&do=blog&id=20736

25minutes, 2019-12-25:
Very nice article. Problem is resolved.

Thanks again SumavisionQ5 & shantigilbert. Merry Christmas! :)

closed 2019-12-25 by 25minutes