EmuELEC/EmuELEC #62: How to configure EmuELEC to sync time from a local NTP server?

25minutes, 2019-12-02:
After I complete building the EmuELEC environment, the s905 device will be moved to a location where it's limited to local network access (no Internet access), and there's a local NTP server. 

I've been searching for the ntp configs in the system, but unfortunately I could find it. Is it defined in a script or somewhere else? Could you please give some hints?

asakous, 2019-12-02:
there is busybox version of ntpd in /usr/sbin  and you need to enable cron service and compile again  . because it's disable by default

MiltonLai, 2019-12-02:
Check this:
```
EmuELEC:/usr/lib/systemd/system # journalctl |grep ntp
Apr 11 11:28:36 EmuELEC kernel: Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
Dec 31 18:00:27 EmuELEC connmand[2752]: ntp: adjust (jump): +155194265.690327 sec
Dec 01 23:48:38 EmuELEC connmand[2752]: ntp: adjust (slew): +0.046881 sec
Dec 02 00:05:43 EmuELEC connmand[2752]: ntp: adjust (slew): +0.013683 sec
Dec 02 00:22:50 EmuELEC connmand[2752]: ntp: adjust (slew): +0.019707 sec
Dec 02 00:39:55 EmuELEC connmand[2752]: ntp: adjust (slew): +0.000914 sec
Dec 02 00:57:01 EmuELEC connmand[2752]: ntp: adjust (slew): -0.030214 sec
```
The configuration file of connmand: /etc/connman/main.conf
```
# List of Fallback timeservers separated by ",".
# These timeservers are used for NTP sync when there are
# no timeservers set by the user or by the service, and
# when UseGatewaysAsTimeservers = false. These can contain
# mixed combination of fully qualified domain names, IPv4
# and IPv6 addresses.
FallbackTimeservers = 0.pool.ntp.org,1.pool.ntp.org,2.pool.ntp.org,3.pool.ntp.org
```
I didn't change that before, maybe you can have a try.

25minutes, 2019-12-02:
Thanks MiltonLai. Problem is solved.

/etc is readonly filesystem, so I choose to configure my local DNS server to resolve 0.pool.ntp.org to my local NTP server. It works.

closed 2019-12-02 by 25minutes