EmuELEC/EmuELEC #63: ES display is not restored after I force PPSSPPSA to run under 720p60hz. How to fix it?

25minutes, 2019-12-05:
It's EmuELEC v3.1 (on s905m) running at 1080p60hz. Most emulators work great at this resolution. Due to performance reason, I'm trying to force PPSSPPSA to run under 720p60hz. 

Below are the changes I made to /storage/.config/emuelec/scripts/emuelecRunEmu.sh. Change 1 (to force 720p60hz for ppsspp):
`
	"psp")
		if [ "$EMU" = "PPSSPPSA" ]; then
		#PPSSPP can run at 32BPP but only with buffered rendering, some games need non-buffered and the only way they work is if I set it to 16BPP
		# /emuelec/scripts/setres.sh 16 # This was only needed for S912, but PPSSPP does not work on S912
		# force ppsspp to run under 720p60hz
		echo 720p60hz > /sys/class/display/mode
		fbset -fb /dev/fb0 -g 1280 720 1280 1440 16
		set_kill_keys "ppsspp"
		RUNTHIS='${TBASH} /usr/bin/ppsspp.sh "${ROMNAME}"'
		fi
		;;
`
Change 2 (to restore 1080p60hz):
`
# Show exit splash
echo 1080p60hz > /sys/class/display/mode
fbset -fb /dev/fb0 -g 1920 1080 1920 2160 32
${TBASH} /emuelec/scripts/show_splash.sh exit
`
The result is, PPSSPPSA can run under 720p60hz, but after I quit it, ES shows only blank screen at the left top corner. It seems it fails to restore 1080p60hz. How can I fix it?

BTW, another minor issue, the kill key (hotkey+start) doesn't work for PPSSPPSA.

shantigilbert, 2019-12-05:
Try this

```
echo "720p60hz > /sys/class/display/mode"
/emuelec/scripts/setres.sh
set_kill_keys "PPSSPPSDL"
```

```
# Show exit splash
echo 1080p60hz > /sys/class/display/mode
/emuelec/scripts/setres.sh
${TBASH} /emuelec/scripts/show_splash.sh exit
```

Note killkeys are dissabled on purpose since you can exit the emulator from the menu

25minutes, 2019-12-05:
Thanks shantigilbert! It works like a charm!

I take further look at /emuelec/scripts/setres.sh, and I'm really curious. On S905, it seems all the script does is the fbset command. Then why it works by calling the script but does't work by running fbset directly? I've verified that either of the following works properly. Mind point out the magic? :)

either 
		echo 720p60hz > /sys/class/display/mode
		/emuelec/scripts/setres.sh
		set_kill_keys "ppsspp"

or

		echo 720p60hz > /sys/class/display/mode
		/emuelec/scripts/setres.sh 16
		set_kill_keys "ppsspp"

and on exit

		# Show exit splash
		echo 1080p60hz > /sys/class/display/mode
		/emuelec/scripts/setres.sh
		${TBASH} /emuelec/scripts/show_splash.sh exit

shantigilbert, 2019-12-05:
Difference is the `16` at the end, setres.sh always uses 32b for s905 even if you supply it with 16

25minutes, 2019-12-05:
Ah, you are right! I missed that part in the script (I'm not an expert on Linux script). 

No worry about killkeys. I'm perfectly fine to quit PPSSPP from its own menu. It's just that when I study the scripts, I see killkeys are set for ppsspp but find it doesn't work. So I thought I should report it. 

I wasn't even know before I study the scripts, that hotkey+start can quit standalone emulators.  If I remember correctly, on RetroPie it only works for RA cores.

Now I have a final question. I remember that some emulator (maybe Drastic, I need verify) can only work properly with 16BPP. So, if for any reason I set the display at 16BPP, is there a method to restore the default 1080p60hz 32BPP?

shantigilbert, 2019-12-05:
Yeah the killkeys was something I wanted to have because of all the standalone emulators that need a keyboard to quit. 

Unless you are running the add-on version, on standalone no need to set 16 anymore, all the emulators work fine at 32bpp

25minutes, 2019-12-05:
Got it. Thank you so much!

closed 2019-12-05 by 25minutes