EmuELEC/EmuELEC #68: Can't get sub folder game when ROMS on USB media

Z0287yyy, 2019-12-22:
Hello,
  When I use ROMS on USB media, the games in sub folder are invisible, such as "\roms\mame\test1\xxxx.zip".
But when it in inner SD card they are visible.

And in old version the games in sub folder are tree struct in ES, but in new version they are flat, I don't think this is a very good change.

Thank you for seeing my chatter.
Best wishes.

shantigilbert, 2019-12-22:
You can configure that with the game options menu (select when inside a category)

Z0287yyy, 2019-12-23:
Ok, I'll get it a try, and feedback then. Thank you.

Z0287yyy, 2019-12-23:
> You can configure that with the game options menu (select when inside a category)

I can't find the menu in ES, could you point it to me, PS, I use the EmuELEC v3.2.

Many thanks.

Z0287yyy, 2019-12-23:
> You can configure that with the game options menu (select when inside a category)

I entry select button in game list, and select "SHOW FOLDERS" = "ALWAYS", but still, neither folder nor games displayed.

Is there other setting?

shantigilbert, 2019-12-23:
Then I have no idea sorry, it should not matter that the folders are on USB

Z0287yyy, 2019-12-24:
Thank you all the same.

closed 2019-12-26 by shantigilbert