EmuELEC/EmuELEC #69: Cannot set timezone

MiltonLai, 2019-12-26:
The system time is 16 hours behind my local time, I found /etc/localtime is pointed to /usr/share/zoneinfo/Mexico/General through /var/run/localtime
```
EmuELEC:~ # ll /etc/localtime 
lrwxrwxrwx    1 root     root            18 Dec 21 00:16 /etc/localtime -> /var/run/localtime
EmuELEC:~ # ll /var/run/localtime 
lrwxrwxrwx    1 root     root            34 Dec 26 02:21 /var/run/localtime -> /usr/share/zoneinfo/Mexico/General
```
But if I change /var/run/localtime, it will be restored after reboot.
I found nowhere to set the timezone after I looked through ES and RA. Is there any way to change it?

Thanks!

shantigilbert, 2019-12-26:
Change it in /emuelec/configs/emuelec.conf

MiltonLai, 2019-12-26:
It works. Thank you!

closed 2019-12-26 by MiltonLai