EmuELEC/EmuELEC #71: request: add the new Xbox One Wireless Adapter driver for linux

rubenjavier, 2019-12-28:
Last week the linux version of the open source driver for the Xbox One Driver Wireless Adapter was released, its also easy to compile

If added to EmuELEC it would be the first retro distribution to have it, and allow to use the wireless adapter to detect Xbox One controllers as regular wired controllers

Here is the repo:
https://github.com/medusalix/xow

Thanks in advance

shantigilbert, 2019-12-30:
It only compiles for newer kernel (only the Amlogic-ng project) But I have no way of knowing if it works as I do not own the dongle.

shantigilbert, 2020-01-17:
IS there a way you can test this? it should be included on the new v3.3

rubenjavier, 2020-01-18:
I have an S912 box and the older wireless adapter. Is this now compiled for my older Amlogic chipset or do we still need the S922 boxes to test?
thanks

shantigilbert, 2020-01-18:
Nope, only for newer chips, the drive does not support older kernels

rubenjavier, 2020-01-19:
Im gonna have to wait till CoreELEC start using mainline kernel, shouldn't be long now, panfrost graphics driver got mainlined recently and its working fine in video and 2D/3D on the T820 of the S912

or maybe somebody with a newer S922 box and a wireless adapter can test it

shantigilbert, 2020-01-27:
I am going to close this issue as the driver is included, albeit not tested.

closed 2020-01-27 by shantigilbert

badkarmaboy, 2020-05-02:
Sorry to revoke this issue, but is the driver really included (also in EmuELEC 3.5)?

I have a XBOX One (S) Wireless Controller, version 1708 and if I plug in the proprietary USB dongle it's not working:

EmuELEC:~ # dmesg
`[    2.003786@1] usb 1-1.2: new high-speed USB device number 3 using xhci-hcd
[    2.128877@1] usb 1-1.2: New USB device found, idVendor=045e, idProduct=02fe
[    2.128883@1] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.128885@1] usb 1-1.2: Product: XBOX ACC
[    2.128888@1] usb 1-1.2: Manufacturer: Microsoft Inc.
[    2.128890@1] usb 1-1.2: SerialNumber: 293059
[    2.130339@1] usb 1-1.2: Unsupported device`

EmuELEC:~ # lsusb
`Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1997:1221
Bus 001 Device 003: ID 045e:02fe Microsoft Corp.
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

EmuELEC:/ # ls -lah /sys/class/input
`drwxr-xr-x    2 root     root           0 May  2 02:09 .
drwxr-xr-x  104 root     root           0 May  2 02:09 ..
lrwxrwxrwx    1 root     root           0 May  2 02:09 event0 -> ../../devices/virtual/input/input0/event0
lrwxrwxrwx    1 root     root           0 May  2 02:09 event1 -> ../../devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.0/0003:1997:1221.0001/input/input1/event1
lrwxrwxrwx    1 root     root           0 May  2 02:09 event2 -> ../../devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.1/0003:1997:1221.0002/input/input2/event2
lrwxrwxrwx    1 root     root           0 May  2 02:10 event3 -> ../../devices/platform/ff808040.meson-ir/rc/rc0/input3/event3
lrwxrwxrwx    1 root     root           0 May  2 02:10 event4 -> ../../devices/virtual/input/input4/event4
lrwxrwxrwx    1 root     root           0 May  2 02:09 input0 -> ../../devices/virtual/input/input0
lrwxrwxrwx    1 root     root           0 May  2 02:09 input1 -> ../../devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.0/0003:1997:1221.0001/input/input1
lrwxrwxrwx    1 root     root           0 May  2 02:09 input2 -> ../../devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.1/0003:1997:1221.0002/input/input2
lrwxrwxrwx    1 root     root           0 May  2 02:10 input3 -> ../../devices/platform/ff808040.meson-ir/rc/rc0/input3
lrwxrwxrwx    1 root     root           0 May  2 02:10 input4 -> ../../devices/virtual/input/input4
lrwxrwxrwx    1 root     root           0 May  2 02:09 mice -> ../../devices/virtual/input/mice
lrwxrwxrwx    1 root     root           0 May  2 02:09 mouse0 -> ../../devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.1/0003:1997:1221.0002/input/input2/mouse0`

EmuELEC:/ # lsmod
`Module                  Size  Used by
nfsv3                  57344  0
nfs_acl                16384  1 nfsv3
nfs                   319488  1 nfsv3
fscache               131072  1 nfs
lockd                 126976  2 nfs,nfsv3
grace                  16384  1 lockd
sunrpc                450560  4 nfs_acl,lockd,nfs,nfsv3
8021q                  36864  0
mali_kbase            499712  3
joydev                 24576  0
ir_lirc_codec          16384  0
lirc_dev               24576  1 ir_lirc_codec
meson_ir               16384  0
rc_core                36864  4 ir_lirc_codec,meson_ir,lirc_dev
overlay                61440  6
wifi_dummy             16384  0
vpu                    53248  0
encoder                57344  0
amvdec_vp9            126976  0
amvdec_vc1             65536  0
amvdec_real            45056  0
amvdec_ports          200704  0
v4l2_common            16384  1 amvdec_ports
videobuf2_dma_contig    20480  1 amvdec_ports
videobuf2_memops       16384  1 videobuf2_dma_contig
v4l2_mem2mem           28672  1 amvdec_ports
videobuf2_v4l2         28672  2 amvdec_ports,v4l2_mem2mem
videobuf2_core         49152  3 amvdec_ports,v4l2_mem2mem,videobuf2_v4l2
amvdec_mpeg4           65536  0
amvdec_mpeg12         106496  0
amvdec_mmpeg4          40960  0
amvdec_mmpeg12         45056  0
amvdec_mmjpeg          32768  0
amvdec_mjpeg           40960  0
amvdec_mh264          163840  0
amvdec_h265           155648  0
amvdec_h264mvc         61440  0
amvdec_h264           147456  0
amvdec_avs             69632  0
amvdec_avs2           212992  0
stream_input          192512  11 amvdec_h265,amvdec_mh264,amvdec_h264mvc,amvdec_ports,amvdec_real,amvdec_vp9,amvdec_h264,amvdec_avs2,amvdec_mpeg12,amvdec_avs,amvdec_mmpeg12
decoder_common        184320  18 amvdec_h265,amvdec_mjpeg,amvdec_mh264,amvdec_mmpeg4,amvdec_h264mvc,amvdec_mmjpeg,amvdec_ports,amvdec_real,stream_input,amvdec_vp9,amvdec_h264,encoder,amvdec_avs2,amvdec_mpeg12,amvdec_avs,amvdec_vc1,amvdec_mmpeg12,amvdec_mpeg4
firmware               28672  18 amvdec_h265,amvdec_mjpeg,amvdec_mh264,amvdec_mmpeg4,amvdec_h264mvc,amvdec_mmjpeg,decoder_common,amvdec_real,stream_input,amvdec_vp9,amvdec_h264,encoder,amvdec_avs2,amvdec_mpeg12,amvdec_avs,amvdec_vc1,amvdec_mmpeg12,amvdec_mpeg4
media_clock            45056  13 amvdec_h265,amvdec_mh264,decoder_common,vpu,amvdec_ports,firmware,stream_input,amvdec_vp9,amvdec_h264,encoder,amvdec_avs2,amvdec_mpeg12,amvdec_avs
amlvideodri            24576  0
videobuf_res           16384  1 amlvideodri
videobuf_core          32768  2 amlvideodri,videobuf_res
videodev              286720  6 amlvideodri,v4l2_common,amvdec_ports,videobuf2_core,v4l2_mem2mem,videobuf2_v4l2
media                  40960  1 videodev
quota_v2               16384  0
quota_tree             20480  1 quota_v2
fbcon                  61440  0
bitblit                16384  1 fbcon
softcursor             16384  1 bitblit
font                   20480  1 fbcon`
EmuELEC:/ #

Will be more than happy to provide more details if requested.

ztoday21, 2020-09-04:
It's not work. i use hardkenel n2 and emuelec 3.8 test version