EmuELEC/EmuELEC #72: confuse about cpufreq/scaling_available_frequencies

keeno1982, 2019-12-30:
S905L can force work max_freq 2.016Ghz on android, but on EmuELEC/CoreELEC only work max_freq on 1.56Ghz, why? 
[image]
[image]

shantigilbert, 2019-12-30:
Are you sure that when you set 2.016 ghz it is actually running at 2.016 ?

keeno1982, 2019-12-30:
@shantigilbert Yes, I can check it work on 2.016 ghz by cpu-z. And then I run some DC emu games like DOA2, it more smoother than force set 1.536 ghz.

keeno1982, 2019-12-30:
[image]

shantigilbert, 2019-12-30:
What DTB are you using ?

keeno1982, 2019-12-31:
gxl_p212_1g.dtb or gxl_p281_1g.dtb

shantigilbert, 2020-01-02:
Sorry I am not sure what is going on, the dtb has the correct info AFAIK

`max-frequency = <200000000>;`

But I am surprised it even works on S905L as the p211 is not in the Linux Kernel so that might be the problem.

keeno1982, 2020-01-02:
@shantigilbert I had decompiled  gxl_p212_1g.dtd and get gxl_p212_1g.dts, if possibly to change some parameters about cpu max_freq and let it work on 2.016 ghz? gxl_p212_1g.dts contains many parameters and I don't know what should I modify. Can you help me to point out it or make a new one?

shantigilbert, 2020-01-02:
Unfortunately I do not know how to make a dtb for a device.

asakous, 2020-01-03:
I am not sure it's related to https://forum.odroid.com/viewtopic.php?t=23044&start=50, but worth to know.

shantigilbert, 2020-01-03:
@asakous that is something I've read before, which is why I asked him if he noticed any difference... he claims yes, but I don't think the CPU is running at 2ghz

shantigilbert, 2020-01-08:
@keeno1982 I am closing this as the link that @asakous shared proves that the CPU doesn't really go over 1.5ghz, if you find any other information feel free to reopen

closed 2020-01-08 by shantigilbert