EmuELEC/EmuELEC #73: Intro video not working

Oberworld, 2019-12-30:
Hi, merry christmas, intro video is not working, seems it's a wrong path in \\EMUELEC\Storage\.config\emuelec\scripts\show_splash.sh.

The lines from 12 to 14:

`GAMELOADINGSPLASH="/storage/.config/splash/loading-game.png"
 DEFAULTSPLASH="/storage/.config/splash/splash-1080.png"
 VIDEOSPLASH="/usr/config/splash/emuelec_intro_1080p.mp4"`

But the intro video is in "/storage/.config/splash/emuelec_intro_1080p.mp4"

My best regards, sorry my poor english.

shantigilbert, 2019-12-30:
Every time I do a release the version goes into the test server, and I personally test it on 9 different devices, and nobody else seems to have this problem. so I am 99% sure this is something on your end, without more information I do not know why this problem is happening. 

Path is not the problem

Oberworld, 2019-12-31:
I did a 3.2 fresh install in my pandora key 7.
If you need more info I'm pleased to help you.
Regards.

shantigilbert, 2020-01-02:
well I see no problem with the intro video, so it is something on your side, not sure what error you are getting, try enabling the option to always show intro video and reboot

shantigilbert, 2020-01-08:
Pk7/s912 seems to have problems with the recent changes to the splash, for the moment I do not intend to fix this, but maybe in the future I will try to find a solution that is suitable for S912 as well, sorry

futxi, 2020-01-11:
A mi tampoco se me ve el video de introducción de pk7, me pasa lo mismo.
Si encontráis alguna manera lo comunicáis. 
Gracias

Oberworld, 2020-01-14:
Thanks for your answer, it's good to know that is not my fault.
I hope you can spend some time in the future to fix this little (but not important) issue.
Again, thanks for your great work.
Regards.

shantigilbert, 2020-01-20:
I think I just figured out a way to make this work, next release will probably have this fixed (if all tests go as I plan).

commit 6ff62c53 (2020-01-20): Use ffplay for intro video and splash

This also adds supports gif splash and fixes #73
closed 2020-01-20 by shantigilbert