EmuELEC/EmuELEC #75: what would be the dtb file for s905M-B 4gb ram processor

mvventura, 2020-01-07:
Please, can someone help me?? i can't boot emuelec

i have a tx9 with s905M-B 4gb ram processor

what would be the dtb file ???

shantigilbert, 2020-01-07:
Read the wiki

closed 2020-01-07 by shantigilbert

YoloDoge, 2021-01-03:
The wiki doesn't provide this info. I'm looking for this info as well

shantigilbert, 2021-01-03:
Like I said, read the WiKi https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install

YoloDoge, 2021-01-03:
I did, like 100 times, but I see no information about S906**M**. Is it supported? If so, which DTB do we need to use for this device? I have been trying several DTB's, but without succes...

shantigilbert, 2021-01-03:
How did you read it 100 times and missed this part? 

```
Note: You can find more information about the DTB that you need by running android and Terminal Emulator type: cat /proc/device-tree/amlogic-dt-id It won't give you the exact DTB but you will get enough information to figure it out. Experiment with things like RAM size (1G/2G/3G/4G) and try a _gibt variant if your box has a gigabit NIC.
```
Please read it 100 times more if needed.

Ghaleon4, 2025-02-03:
Hey shantigilbert.  You are mean-spirited and for no good reason.  

My device does not give me any kind of boot menu or terminal access.  My device uses a '905M' cpu.  None of the DTMs references that CPU name and I don't know  yet if the 'x' in 905 is an _actual_ x or if it represents a variable.

worstcase-scenario, 2025-02-03:
If your device runs android, you have access to the terminal, if you just install an app. If you would read the wiki thoroughly, you should know that and also that you better not insult the man who created Emuelec for free.

Langerz82, 2025-02-06:
try gxl_p281_4g.dtb if it doesnt work it will not boot.

brianw777, 2025-02-06:
> Hey shantigilbert. You are mean-spirited and for no good reason.
> 
> My device does not give me any kind of boot menu or terminal access. My device uses a '905M' cpu. None of the DTMs references that CPU name and I don't know yet if the 'x' in 905 is an _actual_ x or if it represents a variable.

This may help to explain the specs of the S905X SoC. In that article there is a link to some info on the S905M SoC. I was very confused before (still am at times) when I started trying to get Armbian or one of the Elec (EmuElec, CoreElec, LibreElec, etc...) working on my unbranded S905W2 box. The X in S905X isn't just a variable.

[Amlogic S905X Processor Specifications](https://www.cnx-software.com/2016/01/12/amlogic-s905x-processor-specifications/)

fabiofg1982, 2025-04-06:
> Eu fiz, tipo, 100 vezes, mas não vejo nenhuma informação sobre o S906 **M.** Ele é suportado? Se sim, qual DTB precisamos usar para este dispositivo? Tenho tentado vários DTBs, mas sem sucesso...

O meu funcionou com o gxbb_p200_1G_100M_RealtekWiFi , mas apenas com uma versão do emuelec Russo que baixei de um site ruso: https://ollo-store.ru/about_us.html

Ghaleon4, 2025-04-25:
> First of all, it sounds quite like you have a device that is probably using one of those cheap, illegally modified and monetized Asian devices. If that device is able to run android, you have access to the terminal by ust installing a terminal app. Otherwise you can ssh in and open the terminal directly via wifi. If you would have read the wiki thoroughly, you should know that and also that you better not insult the man who created Emuelec for free.

You were correct:  It was an 'illegally modified and monetized Asian device'.  At the time I did not know there was a distinction and was only trying to help a friend that is in prison because it is one of the only gaming devices that they are allowed to use.
In regard to 'reading the wiki thoroughly', and not 'insulting the man who created emuelec for free', kindly suck my balls with the same ferocity that you give to shantigilbert.  : )

worstcase-scenario, 2025-04-25:
I could answer that your friend seems like the one who's the expert in sucking balls in jail, but I won't do that.