EmuELEC/EmuELEC #76: Amlogic-NG: There is a big performance hit if you use Retroarch directly

shantigilbert, 2020-01-08:
There has been a longstanding and significant bug on devices that use the Amlogic-NG build (for S905x2, S905x3, S922x, A311D) for a long time now that will reduce performance in 2 scenarios: 

1.- ES will show screen tearing after loading any game/emulator.
2.- Using Retroarch directly (from the quit menu or by booting directly) will show significant performance drop after exiting a game and loading another one(or the same one) without restarting Retroarch.

On the first scenario it ONLY affects emulation station and does not affect the games at all, so I suggest not to use Retroarch directly.

Video here: https://www.youtube.com/watch?v=F7P7iY2V1AE

At the moment I do not know where this bug is coming from and have no idea how to fix it, but to my knowledge this also affects Ubuntu and other emulation distros for these devices as well.

More general info can be found here: https://forum.odroid.com/viewtopic.php?f=181&t=34827&p=276367

so if anyone has any idea how to fix this bug or has a workaround that we could implement, please let me know. 

In the meantime, if you have any of those devices, use Emulationstation as your front-end the performance drop on Emulationstation does not affect Retroarch or other emulators.

shantigilbert, 2020-02-14:
This is confirmed to be an issue with FBDEV only, as it does not happen on GBM.

ozmafioso, 2020-03-11:
Hello,

I'm using EmuELEC on a S905x3 build and that's what brought me here. I was curious about the issue after loading a game. It's very choppy after loading and quitting out of any game on any core. I would also like to know if there is a solution for this issue. Thanks in advance!

shantigilbert, 2020-03-11:
> Hello,
> 
> I'm using EmuELEC on a S905x3 build and that's what brought me here. I was curious about the issue after loading a game. It's very choppy after loading and quitting out of any game on any core. I would also like to know if there is a solution for this issue. Thanks in advance!

Like the issue says:

"In the meantime, if you have any of those devices, use Emulationstation as your front-end the performance drop on Emulationstation does not affect Retroarch or other emulators."

ozmafioso, 2020-03-11:
> Like the issue says:
>
> "In the meantime, if you have any of those devices, use Emulationstation as your front-end the performance drop on Emulationstation does not affect Retroarch or other emulators."

Right, I agree. I was just hoping there was a fix for the front end, to fix the choppiness after loading a game. It’s not a huge deal, I just love the smoothness in the FE. The actual games themselves still run phenomenal as mentioned. I appreciate your time!

shantigilbert, 2020-04-05:
Update on this issue, latest release v3.5 fixes the sluggishness on Emulationstation, although the Retroarch part still has the bug.

george291214, 2020-05-22:
Usaba emuelec en un tv box a95x f3 air con s905x3 y todo iba bien y un dia ya no encendio el tv box y solo enciende un foquito azul dentro y no se ve nada en pantalla alguien sabe cual podria ser el problema ?;
[image]
[image]

shantigilbert, 2020-05-22:
@george291214 eso no tiene nada que ver con EmuELEC, ni es el lugar indicado para pedir ese tipo de ayuda.

meeceleroy, 2020-11-22:
Does all of them chipsets use the same Emuelec download file.

Matteokaiat86, 2021-01-06:
Ciao ragazzi io ho un tanix 5 plus amlogic s905x3 ma non riesco a fare il flash EmuELEC-Amlogic-ng.arm-3.9-Generic.img.gz poi copiando ed incollando l'altro per il 4gb di ram eseguo il boot  dall'app libreelec ma non mi avvia emuelec mi si inchioda sul boot anche per un ora e più cosa può essere?

spleen1981, 2022-01-05:
Trying to develop a simple [retroarch addon for coreelec](https://github.com/spleen1981/retroarch-kodi-addon-CoreELEC), and facing several issues with mali_fbdev, including this long-standing one.
I think problem is in [egl_destroy](https://github.com/libretro/RetroArch/blob/d13f325216628031f2937b54e854c0922f4b5a54/gfx/drivers_context/mali_fbdev_ctx.c#L64) call in mali_fbdev context driver source.
Findings:

- Each time there is a change in video settings (e.g. monitor index change, useful as quick check) resolution etc (including launching a game from retroarch GUI) following cycle seems to occur:
1. `gfx_ctx_mali_fbdev_destroy` hence `egl_destroy`
2. `gfx_ctx_mali_fbdev_init`
3. `gfx_ctx_mali_fbdev_set_video_mode`

   while on first GUI loading (or launching a game from CLI directly) step 1 does not apply.
That seems to be the reason why at the beginning GUI works good as well as direct CLI game launching.
- Testing on S905X3, if initially I had 60fps in the GUI; switching the monitor index fps drops to 33.
- No further fps drops occur with additional monitor index changes, btw.
- As a dirt check, tried to build with  `egl_destroy` commented out from `gfx_ctx_mali_fbdev_destroy`.
   In this case no fps drop can be observed in the GUI switching monitor index. Also performance in game launched from GUI are much better, even if while playing fps slowly drops to 45-50 fps.
   Of course this is not a solution, as resources will reasonably leak for each missed destroy cycle above (infact after few of them the screen freezes, though no error is printed out), but it is a strong indication that something is wrong with `egl_destroy`.

I guess a deep dive into sources behind this function is needed...

shantigilbert, 2022-01-05:
Thank you for taking a look at this, you seem to be  on the right direction as to why this is happening. To add more information, this issue also used to happen on Emulationstation, in a very similar fashion as you mentioned, when the screen changed the FPS dropped, so this might be an issue with the mali blobs themselves, but if a workaround is found, I am happy to use it

spleen1981, 2022-01-10:
Ok, I think I found a solution, it seem to run flawlessly now.
Check #789 or try the retroarch binary included [here](https://github.com/spleen1981/retroarch-kodi-addon-CoreELEC/releases/download/v1.2.0/script.retroarch.launcher.Amlogic-ng.arm-v1.2.0.zip).
Details are given in the commit message above.
Before trying to merge upstream I will take some time to get additional feedbacks.

shantigilbert, 2022-01-10:
Fixed! amazing work

> Ok, I think I found a solution, it seems to run flawlessly now. Check #789 or try the retroarch binary included [here](https://github.com/spleen1981/retroarch-kodi-addon-CoreELEC/releases/download/v1.2.0/script.retroarch.launcher.Amlogic-ng.arm-v1.2.0.zip). Details are given in the commit message above. Before trying to merge upstream I will take some time to get additional feedbacks.

closed 2022-01-10 by shantigilbert

shantigilbert, 2022-01-10:
FYI: I opened an issue in RA a while back https://github.com/libretro/RetroArch/issues/10918 so if you want to submit a PR  you can close it.

spleen1981, 2022-01-14:
merged upstream in retroarch, patch can be deleted and retroarch bumped to f93548a or later.

shantigilbert, 2022-01-14:
> merged upstream in retroarch, patch can be deleted and retroarch bumped to f93548a or later.

Thanks for the heads up I will update it.

PR #789 (2022-01-10): mali_fbdev fix for fps drop after egl_destroy
Currently each time a screen resolution or setting change occurs, fbdev_destroy, fbdev_init and fbdev_set_mode are called in sequence, trying to destroy context and surface (though context pointer seems to remain unchanged) and create again both of them.
However it seems something is wrong with egl_destroy, as after the call fps drops from 60 to 33fps in GUI with huge performance impact, at least with libmali.
Philosophy is changed with this commit to avoid destroying and creating context and surface each time (creation only occurs in fbdev_init an [... cut]