EmuELEC/EmuELEC #78: gasia-co-ltd-psr-gamepad

joln, 2020-01-11:
Hi !
EmuELEC:3.2

Hoping for help….recently bought PS3 controllers ( a PS3 controller clone they are not Sony). EmuELEC labels them as Gasia when trying to configure. 

**Controller used: Gasia Co.,Ltd PS(R) Gamepad**

Use lsusb:
Bus 001 Device 013: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller

I have gone through the norm but can only get the controller to work via USB and not Bluetooth. Anyone have success with pair similar ones?

-------------------------------
someone has same problem in RetroPie，and it run！！！

https://blog.petrockblock.com/forums/topic/gasia-co-ltd-psr-gamepad/

-------------------------------

But in EmuELEC 

– tried this found in another post:
git clone https://github.com/supertypo/qtsixa.git
cd qtsixa/sixad
make  
sudo mkdir -p /var/lib/sixad/profiles
sudo checkinstall

but EmuELEC can't make
 -sh: make: command not found

Please help if you can!

shantigilbert, 2020-01-12:
DS3 Gamepads come preconfigured on EmuELEC but DS3 controllers have all sorts of problems, I imagine clone gamepads are even worse, if it does't work out of the box I have no idea how to fix this, sorry, I do not have a DS3 Gamepad original nor clone.

joln, 2020-01-12:
maybe qtsixa can add in EmuELEC?

shantigilbert, 2020-01-12:
tried to compile it but it error out, I really don't want to move too much on the DS3 blindly, so unless someone else would like to do this I won't add it like this, sorry.

qacjava, 2021-01-24:
Hi !  Friend ,I'm sorry for read this Q&A now 
my DS3 Gamepads  also is the same model.
please take qtsixa  add in  new version EmuELEC  
thank you and pray！thank you very much!

qacjava, 2021-01-24:
I know you tried to compile it but it error out .
 maybe this qtsixa can add in EmuELEC?
http://qtsixa.sourceforge.net/  
please  hope  you can add it.

Thank you from the bottom of my mind

bleemvip, 2021-01-24:
You also want to install the plug-in

bleemvip, 2021-01-24:
我也是这个手柄，希望在新版本的EMUELEC上使用手柄插件。

bleemvip, 2021-01-24:
I am also the game controller, and I hope to use the plug-in on the new version of emuelec.

qacjava, 2021-01-28:
@shantigilbert please~ Thank you from the bottom of my mind

Langerz82, 2023-12-14:
Try this:
https://github.com/EmuELEC/EmuELEC/pull/1201

Or build from the latest dev version and re-test.

This issue is very old so if it's still an issue please re-open a new issue.

closed 2023-12-14 by Langerz82