EmuELEC/EmuELEC #79: X95max WIFI and Bluetooth don’t work

Carlosaleqwe, 2020-01-11:
Hi I just bought a X96max, 4G/32G  The problem now is that WIFI and Bluetooth don’t work
This is the run log

https://pastebin.com/jkcbPtUV

i trywith a bluetooth adapter to connect an ipega controller, but always have to pair it and in emulationstation can't map the buttons, just recognize the controller as a keyboard

shantigilbert, 2020-01-12:
If you are sure you are using the correct DTB then no idea, those are 2 things I do not use, these devices all have different BT/WiFi chips so its hard to even know why. 

as for WiFi try connecting to Wifi over Retroarch.

Carlosaleqwe, 2020-01-12:
g12a_s905x2_4g.dtb
i tried on retroarch without results

Carlosaleqwe, 2020-01-14:
Cpu model Amlogic s905x2 (REV B) 
g12a_s905x2_4g.dtb

shantigilbert, 2020-01-18:
If You don't get the network list over Retroarch then either your WiFi chip is not supported or the DTB you are using is incorrect. Check your WiFi chip by opening your device.

closed 2020-01-18 by Carlosaleqwe