EmuELEC/EmuELEC #87: Failing to build EmuElec

jonasmac16, 2020-01-22:
Hi,
I have setup a VM with Ubuntu and followed the instructions of how to build EmuElec but I get the following error:
```
obj-unix/release/./libretro-common/audio/resampler/drivers/sinc_resampler.o:sinc_resampler.c:function resampler_sinc_process_neon: error: undefined reference to 'process_sinc_neon_asm'
obj-unix/release/audio/drivers_resampler/cc_resampler.o:cc_resampler.c:function resampler_CC_upsample: error: undefined reference to 'resampler_CC_upsample_neon'
obj-unix/release/audio/drivers_resampler/cc_resampler.o:cc_resampler.c:function resampler_CC_downsample: error: undefined reference to 'resampler_CC_downsample_neon'
obj-unix/release/./libretro-common/audio/conversion/s16_to_float.o:s16_to_float.c:function convert_s16_to_float: error: undefined reference to 'convert_s16_float_asm'
obj-unix/release/./libretro-common/audio/conversion/float_to_s16.o:float_to_s16.c:function convert_float_to_s16: error: undefined reference to 'convert_float_s16_asm' 
```

Attached full error log:

[error_make.txt](https://github.com/EmuELEC/EmuELEC/files/4099656/error_make.txt)

Is this a known issue?

shantigilbert, 2020-01-22:
It is not a known issue, never done a VM to compile it, maybe you are missing a dependency? what command line did you use?

jonasmac16, 2020-01-22:
I also had the same issue under Manjaro which runs directly on my laptop. I made sure to install all dependencies stated in the Readme 

Both command
```
git checkout EmuELEC-aarch64
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image
```
and 

```
git checkout EmuELEC-aarch64
 PROJECT=Amlogic-ng ARCH=arm DISTRO=EmuELEC make image
```
result in the above error

shantigilbert, 2020-01-22:
You are trying to compile the 64bit arm version (which btw is not fully working) using an ARM arch...

If you really want to compile the 64bit version try 

PROJECT=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image

jonasmac16, 2020-01-23:
Fair enough, should have payed attention to that detail. I will try using the master branch and report back. I just did `git checkout EmuELEC-aarch64` as the instruction suggested to checkout another branch other than `master`.

shantigilbert, 2020-01-23:
Ohh you are right, I need to update that instruction it used to be emuelec but I changed it a while ago to master

shantigilbert, 2020-01-23:
Fixed readme

closed 2020-01-23 by shantigilbert