EmuELEC/EmuELEC #92: Color is off/wrong

nickvasallo, 2020-02-02:
Hello, I installed EmuELEC and the color is off, everything is yellow/turquoise. Is the DTB wrong?

nickvasallo, 2020-02-02:
Using EmuELEC 3.3 and g12a_s905x2_4g.dtb for this box: https://www.amazon.com/Android-EstgoSZ-Decoding-5-0G100M-Ethernet/dp/B074C8XLCR

shantigilbert, 2020-02-02:
On the first partition of the sd card there is a file called config.ini, fine this line 
```
# use_rgb_to_yuv=0
```
and un-comment the line,  so its like this
```
use_rgb_to_yuv=0
```
Insert SD card back to the device and it will look normal

nickvasallo, 2020-02-02:
I ended up using 3.2 and it worked! Gonna stick with 3.2

shantigilbert, 2020-02-02:
Ok, but that fix works on 3.3 just in case you want to upgrade later :)

closed 2020-02-02 by shantigilbert

nickvasallo, 2020-02-02:
thanks man! noted