EmuELEC/EmuELEC #95: N64 roms corrupt when change resosolution

taso-tac, 2020-02-06:
Hi! By default N64 runs in 320p but if i access to retroarch during play and change resolution to 720p or 576p  and reload rom, it´s looks great and run perfect while not turn it off. Once i close it the game and return some  graphicals disappears: Expansion pack picture, background, cursor, etc. Is impossible to me come back to restore functionalitty, no other changes works: Core, resolution, restart sistem, screen size, etc.
 -  Do you know whats happens, is possible this emulator load some cache and it´s corrupt???
 -  Where is the configuration file??? Can i delete it to back to default?

This problem is permanent, i only change resolution by game. Rest of games run ok. Now four game runs corrupt.

Thx!

Great work!!!!

shantigilbert, 2020-02-08:
Never heard of this happening, I have no idea what could it be

shantigilbert, 2020-03-11:
I cannot reproduce this whatsoever on all my devices, this sounds to me like a bad SD. Can you reproduce and give the exact steps?

closed 2020-04-05 by shantigilbert