EmuELEC/EmuELEC #97: Is add-on version not updated?

siyiye, 2020-02-08:
It stays at 3.2.....

Raulval82, 2020-02-08:
Hi, I'm also interested in this, I just discovered that I can keep my kodi installation with emuelec and install the coreelec addon, it is amazing. The updates are just for the image version?

shantigilbert, 2020-02-08:
there are not many changes to the addon yet, so no reason to bump it really, but updates are coming mostly just to use the newest emulator versions

siyiye, 2020-02-09:
> there are not many changes to the addon yet, so no reason to bump it really, but updates are coming mostly just to use the newest emulator versions

How can I upgrade to the latest simulator in 3.2?
THX!

Sheredeka, 2020-04-25:
i'm try create addon:
[root@mc EmuELEC]# ./emuelec-addon.sh
Building EmuELEC KODI add-on for CoreELEC:

DISTRO=EmuELEC
PROJECT=Amlogic
ARCH=arm
VERSION=3.5
GIT_BRANCH=master

Working in: /root/EmuELEC
Temporary project folder: /root/EmuELEC/emuelec_addon_workdir/2020-04-25_192030

Target zip: /root/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher/script.emuelec.Amlogic.launcher-3.5-Amlogic.zip

Removing old add-on at /root/EmuELEC/repo/3.5
Folder '/root/EmuELEC/repo/3.5' exists.
Created folder '/root/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher'
Created folder '/root/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher/resources'

Building packages:
        **retroarch (failed)
Error building package 'retroarch'!**

estebanjnr, 2020-05-03:
Same output as @Sheredeka :

esteban@esteban-lenovo:~/Documentos/Personal/Amlogic/emuelec/EmuELEC$ ./emuelec-addon.sh
Building EmuELEC KODI add-on for CoreELEC:

DISTRO=EmuELEC
PROJECT=Amlogic
ARCH=arm
VERSION=3.5
GIT_BRANCH=master

Working in: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC
Temporary project folder: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/emuelec_addon_workdir/2020-05-02_163009

Target zip: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher/script.emuelec.Amlogic.launcher-3.5-Amlogic.zip

Removing old add-on at /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5
rm: no se puede borrar '/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher/resources': Permiso denegado
Folder '/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5' exists.
Folder '/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher' exists.
Folder '/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/repo/3.5/script.emuelec.Amlogic.launcher/resources' exists.

Building packages:
	retroarch (failed)
Error building package 'retroarch'!

I'm on Ubuntu 19.10 x86_64

artemisoftnian, 2020-05-12:
@estebanjnr @Sheredeka 
To build the add-on version:
First build the images using: `PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image` or `PROJECT=Amlogic-ng ARCH=arm DISTRO=EmuELEC make image`  or bouth.  Be patient it took a while at least on my side.

Second and after the image build compleates.  Run `./emuelec-addon.sh` .  You can comment the file leaving out the version of the build you not going to use, if any. (last two lines on the .sh file)

estebanjnr, 2020-05-12:
Thanks @artemisoftnian 

I tried to build the image but I get an error while the process is building PPSSPP.
I'm getting this in the terminal output:

-- Up-to-date: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/PPSSPPSDL-310e1e4f18047ba602096dac4d451bec35834070/.armv8a-libreelec-linux-gnueabi/assets/flash0/font/ltn1.pgf
-- Configuring incomplete, errors occurred!
See also "/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/PPSSPPSDL-310e1e4f18047ba602096dac4d451bec35834070/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeOutput.log".
See also "/home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/PPSSPPSDL-310e1e4f18047ba602096dac4d451bec35834070/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeError.log".
FAILURE: scripts/install PPSSPPSDL has failed!
[346/416] [FAIL] install PPSSPPSDL

The following logs for this failure are available:
  stdout: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/346/stdout
  stderr: /home/esteban/Documentos/Personal/Amlogic/emuelec/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/346/stderr

parallel: This job failed:
package_worker 4 346 416 'install PPSSPPSDL'
Parallel build failure - see log for details. Time of failure: Tue May 12 16:59:57 CDT 2020
[stdout.txt](https://github.com/EmuELEC/EmuELEC/files/4618724/stdout.txt)
[CMakeError.log](https://github.com/EmuELEC/EmuELEC/files/4618729/CMakeError.log)
[CMakeOutput.log](https://github.com/EmuELEC/EmuELEC/files/4618731/CMakeOutput.log)
make: *** [Makefile:12: image] Error 1

[CMakeError.log](https://github.com/EmuELEC/EmuELEC/files/4618736/CMakeError.log)

[CMakeOutput.log](https://github.com/EmuELEC/EmuELEC/files/4618738/CMakeOutput.log)

[stderr.txt](https://github.com/EmuELEC/EmuELEC/files/4618743/stderr.txt)

[stdout.txt](https://github.com/EmuELEC/EmuELEC/files/4618744/stdout.txt)

I took a look at logs too but I don't know how to find what I'm missing.

What kind of system are you building on?

artemisoftnian, 2020-05-13:
@estebanjnr  
Hola, esteban!! yep Spanish Too!!! Well inferring cause the Documentos part on log... LOL! 
Anyways I'm still understanding all about this too! So I just imagine maybe a missing library! Tal vez!

I successfully build mine using Ubuntu (Focal) (Latest)

estebanjnr, 2020-05-13:
Que onda @artemisoftnian !? Yeah I'm fairly new to all of the *elec stuff too. How do I get to the Discord Channel? I may be new to that too :). 
I'm gonna take some time to look again to the logs and see what I can dig out. 

Saludos hemano!

lzivadinovic, 2020-05-20:
I've tried to build image for Odroid-N2 (plugins also fails with this step) and it looks like script cant locate media_modules-aml. I've looked on archive libreelec, and it looks like they removed that folder from their servers. Any tips and tricks?

shantigilbert, 2020-05-21:
Please use the dev branch to build the addon, as the master branch is only updated on each release

estebanjnr, 2020-05-26:
I installed Ubuntu Server 20.04 in Virtualbox and built it there successfully (image and add on). If using this approach I suggest to give at least 100gb to the virtual drive. The building takes lots of space.
Do you know if it is possible to only build the image por a specific device? In my case I only would want to build the generic  image.

shantigilbert, 2020-06-28:
New one coming soon e2cb4c276260c2cc8aafd83e5f3a376af69826f8

closed 2020-06-28 by shantigilbert