EmuELEC/EmuELEC #98: eMMC with microSD card

blinciki-lol, 2020-02-08:
Hi, 

I've got an Odroid N2 and wanted to install EmuElec, but my USB adapter doesn't work, so I couldn't install anything and or make any tests.

Could you please let me know, if it's possible to use EmuElec on the eMMC card and the roms on the microSD card? If so, how can I do it?

I've read a guide about how to set it up for batocera, but wanted to try your distro first after reading more recommendations on the web :)

shantigilbert, 2020-02-08:
You should be able to do that, just follow this WiKi entry https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-USB-media and replace the USB with your SD.

blinciki-lol, 2020-02-08:
Thanks for your help and for your work with distro. I'll give it a try, and provide feedback later.

closed 2020-02-08 by blinciki-lol