EmuELEC/EmuELEC #99: Front Display Support?

FairOh86, 2020-02-08:
I recently purchased an S905x3 device with a front display.  While running EmuELEC 3.4 the LED Front Display always shows "boot".
I would like to see if support can be implemented.

I believe this would be the relevant source:
https://github.com/arthur-liberman/linux_openvfd/

shantigilbert, 2020-02-08:
I am sure it could be implemented (and cool!), unfortunately I have no device with a display so I would just be doing everything blind :( 

will look into it anyway

BaltasarParreira, 2020-04-24:
@shantigilbert 
I have a s905x3 box with display, so if it helps I can run any test you might need. ; )

BaltasarParreira, 2020-05-19:
Steps to do for using VFD/display on your TV box:
1- Go to https://github.com/arthur-liberman/vfd-configurations and identify your box.
2- Grab the .conf file for your box and rename it to vfd.conf
3- Put the renamed file here: 
`/storage/.config/vfd.conf`
or using the shares at `\\<your box IP>\Configfiles\vfd.conf`
4- Enable the OpenVFDService like this:
`systemctl enable openvfd`
5- Start the OpenVFDService like this:
`systemctl start openvfd`
6- Reboot the box.

If the vfd.conf you choose is the correct for your box you should start seeing the internal current time on the display, if not try to choose another and reboot again. 
Do this until you find the one that makes the time displaying correctly.

snakect77, 2020-05-22:
thanks with my box t95-max-plus-s905x3 it worked immediately

shantigilbert, 2020-05-22:
Thanks @BaltasarParreira for the find! this can now be closed

closed 2020-05-22 by shantigilbert