EmuELEC/EmuELEC #104: Can't turn on gt-king without disconnecting & reconnecting power cable

ChronicAC, 2020-02-14:
The remote that came with the gt-king SN:B922 box does not work in EmuELEC.
just looking for a way to turn on the box without manually disconnecting & reconnecting power cable, everything else I do from the controller which works great.. 

I'm using the standalone EmuELEC. 
Can't figure this out for the life of me... all the info/remote files I can find on this is all about CoreELEC and related to kodi, and the remote.conf files provided by CoreELEC does not seem to fit in this build, no such path or files?

The only thing I can find in EmuELEC is a file called: odroid-amremote.conf
I can see that this file is for Odroid remote, I guess it should work in Odroid boxes...
is there any way to edit it for the beelink regular remote (black & orange no air mouse)
or any other way to turn on the box will be very appreciated.

if there's no way maybe I can buy a remote that will work?

A big thank you to shantigilbert its the best thing that can happen for this buggy box.

shantigilbert, 2020-02-14:
Ir remotes are not really  implemented on emuelec. I did not want to go through the hassle of doing all of the configurations for just one button. I will add it to my todo list, but I cannot promise anything.

cpu123, 2020-03-13:
Yeah I've noticed this problem with my gt-king and the RF remote/airmouse. Once you've turned it off, it's impossible to turn it back on without pulling the power, unlike in coreelec.

Bit of a pain trying to explain it to my kids!

howels, 2020-03-13:
Yeah, this is the same for my T95 max+ but I figured that this was probably implemented by some weird 3rd party driver for the bIR receiver.

Fugoner, 2020-04-09:
La única solución que puedo llegar hacer es ponerle un interruptor al cable de corriente. Porque inalambricamente creo que no hay opción.

xSmithfieldx, 2020-05-11:
Had the same issue and I found this power cord on Amazon with a button to turn it on. Works great! 

Cordinate, White/Gray, Designer Switch Plug https://www.amazon.com/dp/B07KFMNJLH/ref=cm_sw_r_sms_apa_i_-3jUEbGKM05F6

coach1988, 2020-05-24:
I was experiencing the same under coreelec,got around it by using suspend mode instead (though that seems to break the Bluetooth/wifi sometimes,have to investigate that further) or using CEC,but I think both options are not available via the GUI here,though it perhaps could be done via cli.

shantigilbert, 2020-06-15:
Remote control support has been added to EmuELEC v3.6,  you will need to set it up the same way you don on CoreELEC

closed 2020-06-15 by shantigilbert