EmuELEC/EmuELEC #117: Screen rotated and small

lbernstone, 2020-02-27:
I am building from source for a Amlogic-ng.  I get a working system, but the graphics are rotated -90 and about 320*480 in the lower left corner.  The 3.4 distro works correctly.  I'm guessing this is the correct layout for Odroid Go Advance.  Which file has the logic that controls this?  It does not seem to be identifying the device properly.  Do I need to put a DEVICE option in my make command?

shantigilbert, 2020-02-27:
Please check that the emulationstation service does not have the remaining of the odroidgoadvance setup. I am going to push a new update to that package that properly cleans the service later today

shantigilbert, 2020-02-27:
Commit 9df6065af0a2ce6b4e46c9a8aac0b38457749a6d should fix this issue

closed 2020-02-27 by shantigilbert