EmuELEC/EmuELEC #118: advance mame with xbox360 wireless controller

dewenhsu, 2020-02-29:
Odroid N2 + emuelec 3.5 test + xbox360 wireless controller + usb keyboard
I can use keyboard 6 and 2 to insert coin and start game
the controller can use direction pad, but fire bomb ( A and B button ) is not working.
the keyboard direction and A B ( ctrl and alt ) works.
I wish to call advmame config menu by press keybard "tab",
but it is not work.
Is there a way to call the advmame config menu so I can map the controller ?
thank you !

shantigilbert, 2020-02-29:
What happends if you press the guide button? on the controller, or the L3 or R3 ?

dewenhsu, 2020-03-01:
guide button no response. LT RT and all possible button no response.
but if I use keyboard 6 and 2 to enter game, the direction pad is working.
there is no L3 or R3 for xbox360 wireless controller.
thank you.
there is a config file for advmame ( advmame.rc )
it seems can add config for the controller mapping, but I still can not get success.

shantigilbert, 2020-05-26:
I can confirm that there was an issue with advmame, it will be resolved on the upcoming version.

dewenhsu, 2020-05-26:
thank you very much !

shantigilbert, 2020-06-08:
Can you replace this script `/emuelec/scripts/set_advmame_joy.sh` with this one

https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/config/emuelec/scripts/set_advmame_joy.sh

and also add this `device_joystick raw` to `/storage/.advance/advmame.rc`

And test to see if your gamepad is now working?

dewenhsu, 2020-06-09:
Thank you !
It works. Now I can press keyboard "Tab" to call advmame menu, and config the controller.

shantigilbert, 2020-06-15:
Fixed on v3.6

closed 2020-06-15 by shantigilbert