EmuELEC/EmuELEC #119: xbox360 wireless controller and random video / slideshow screensaver

dewenhsu, 2020-02-29:
Odroid N2 + xbox360 wireless controller.
when I set screensaver to random video or slideshow,
usually I can press direct pad "right" to skip to next video/slide,
but when I press it, it jump back to ES.
can it be fixed ? thank you !

If I press "start", it works to launch the game which play the video/slideshow.

shantigilbert, 2020-02-29:
it is working as it should, its not a slideshow, its a screensaver, its supposed to exit when you press a button. when you say usually what do you mean?

dewenhsu, 2020-03-01:
for example, if I set video screensaver to 30 second, 
if I watch it for 15 second and press "strart" button, I can enter the game.
if I watch 15 second and press "right" direction pad, I can skip this video and jump to next video.
( you can try it on the Odeoid Go Advance )
if I press other button or pad except "start" or "right" direct pad, it will jump back to ES.
My problem is when I press "right" direction pad, it will jump to ES, not skip to next video.
thank you.

dewenhsu, 2020-03-11:
I found if you press “select” button, you csn skip to next video.

shantigilbert, 2020-03-11:
I will close this issue now.

closed 2020-03-11 by shantigilbert