EmuELEC/EmuELEC #128: Cores Missing and Retroarch settings gone after reboot

dickyboy4444, 2020-03-06:
Hi all,
           have installed emuelec latest on an s10 r-tv 912 android box.  Boots up ok.  Upload snes roms via smb and they appear ok.  When I try to boot rom it does nothing.  I add cores via smb and then it starts working.  After reboot all cores are missing and any downloads in retroarch like shaders are also gone.  What am I missing.

shantigilbert, 2020-03-11:
I can't understand your issue. try re-flashing the SD card? also do not use etcher, use win32diskimager

flabla, 2020-04-12:
@shantigilbert hello , i have a problem, i add Emuelec on t95 max+ android box, everything works fine except Neogeo and Mame, how i can make them work ? i added all roms as explained, snes, nes, sega, etc,etc working ok.....Mame and Neogeo nope. Please i need some guidance

shantigilbert, 2020-04-12:
@flabla don't hijack issues, look in emuelec.log

closed 2020-04-12 by shantigilbert