EmuELEC/EmuELEC #129: Can't boot emuElec on T95 Max Plus

ValentinLaTortue, 2020-03-08:
Hello everybody,

I've bought a T95 Max Plus from [amazon](https://www.amazon.fr/Android-Amlogic-Quad-Core-Support-Bluetooth/dp/B082Q331ZZ/ref=sr_1_4?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=R2O3QWOY3Y3S) (Web page is in French) and I'm trying to install emuElec on it. I have a 32GB micro SD and I've tried every single dtb file from both the ng and classic images. I really don't know what I could be doing wrong.

Every time I plug the microSD card and boot the device, it goes straight to the android OS. It seems like the box doesn't even see the microSD. But it still creates Android and LOST.DIR folders.
I've tried to locate a potential reset button but I haven't found any.

I hope somebody can solve my problem, as I really want to use the device for my arcade cabinet.
Thanks for your time.

shantigilbert, 2020-03-08:
Please read the WiKi https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install

ValentinLaTortue, 2020-03-08:
I did, and I followed every single steps at least 10 times. I even followed a tutorial on how to install emuElec on my specific box. 
I've looked through the internet and from my understanding, the box I'm using has a S905x3 processor, but every dtb associated ended up not working at all.

Andromydous, 2020-03-08:
Same here. Mine is by Transpeed. https://www.amazon.com/gp/product/B06XPDN8RY/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
I have spent several hours attempting this with no result. With or without reset button, my box just hangs at the "www.shinings-tech.com" screen (preboot I think). Power off, take out sd card, and power on the box works just fine. Either I'm overlooking something minor and obvious or it just can't be done on this box. The box has s905x3 and 4 gb of ram.

carlclarke, 2020-03-08:
I have the same issue, I watched a youtube video by ETAPrime about using the T95 Max+ with EmuELEC and thought it would be fun. Ordered the same device and tried it out today, no boot from SD, no boot from USB. I went back a version to 3.3 and it would not boot. I tried making the SD and USB disks using balena etcher, rufus and even the new raspberry pi imager. Also I tried removing all partitions from the SD card (although the imaging app should do that anyway). I don't really know what the issue is but the latest balena etcher was failing too (Windows 10 v1903) so I am just wondering if 1903 might have something to do with the problem. If I get free time soon I will try an older Windows version and Linux. Could be a 'red herring' but I feel (without any evidence) it has something to do with the SD card.

shantigilbert, 2020-03-08:
The reset part is something that was missing from the video from ETA and its a very important step, you need to hold reset while you power on the device. In case your device does not have a reset button, install an android app called "reboot to coreelec" (or reboot to libreelec) and with the SD inserted use the app to reboot.

closed 2020-03-08 by ValentinLaTortue

shantigilbert, 2020-03-08:
Was this resolved?

howels, 2020-03-08:
The reset button on the T95 max+ is inside the headphones jack. Put a matchstick or toothpick in there and press the button while you power the box on. Only needed for the first boot.

Andromydous, 2020-03-08:
Yeah well, I've tried the toothpick way (works as long as there isn't an sd card in slot) and I've tried to "reboot to libreelec" way. Neither of them work. Not sure the extra little bit of power is worth this. I was emulating just fine before all of this. I thought "Closed" meant resolved. For one other person and myself, it's not resolved.

shantigilbert, 2020-03-08:
You either have the incorrect DTB or your SD flash is corrupted

carlclarke, 2020-03-08:
I am happy - found a stick, found the reset switch and it worked. Earlier I put on my glasses to look for a reset switch but never thought to look inside the line out jack. Mr ETA 👎, he made it look too easy and he even mentioned in the video that this device didn't have/need a reset button, the devil is in the detail. Thanks @shantigilbert just needed the right instructions 👍

Andromydous, 2020-03-09:
I wish it was that simple, @shantigilbert . The sd card works fine. Being a 4gb RAM and S905x3 there are only 2 choices of dtb files. I tried both. To clarify, I put one one there and renamed it to dtb.img and it didnt work. I then deleted that file and copied over the other one and renamed it to dtb.img, which didn't work either. I am really starting to lean towards the problem being the box itself. I've even tried the website "www.shinings-tech.com" and it comes up as being non-existent. There is literally nothing on the internet involving my box specifically outside of buying it from amazon. Maybe it's a "generic" version of another brand, but I have no way of knowing.

shantigilbert, 2020-03-09:
If your use windows, Make sure your windows is not hiding extensions for the dtb.img

Also if this is a generic device as you say, make sure the box actually is Amlogic, had many users with the same problem just to find out the device is actually another cpu.

Andromydous, 2020-03-09:
I used Aida64 (whatever it's called) to verify that it's an Amlogic s905x3 and how much ram it had. I'll look into the file extension issue. I did make sure to use the ng-generic-4gb and then the ng-generic-4gb-1gbit. I thought about using the maker specific download files, but I'm not sure if it's worth the effort. I, also, wanna say "thank you" for the response.

Andromydous, 2020-03-10:
I reached the break limit today. The seller (supposedly Transpeed) sent me a link to download "firmware update". It was downloading at a fast speed of 120k/s, seemed a little fishy, and ended up being just an update for the Kodi app. I'm getting a refund from Amazon and going to get a box that's well known, has a lot of info online, and even have it's own website.

DKAngel2000, 2020-03-10:
> I reached the break limit today. The seller (supposedly Transpeed) sent me a link to download "firmware update". It was downloading at a fast speed of 120k/s, seemed a little fishy, and ended up [...]

umm download terminal and type in reboot update with the sd card in and it should atleast boot, mine doesnt have a reset button to hold and thats how i get to do mine

Andromydous, 2020-03-10:
@DKAngel2000 I appreciate the help. For reasons already listed above, I've already started the refund process. It's not an operator failure and it's not an issue with the EmuElec. It's a multi-layered issue with the box and the company.

snakect77, 2020-04-24:
I found the same problem with the same box t95max + amlogic s905x3, using apk aida64 mi from model tvbox franklin device franklin platform
can anyone start emuelec on this device?

howels, 2020-04-28:
> I found the same problem with the same box t95max + amlogic s905x3, using apk aida64 mi from model tvbox franklin device franklin platform
> can anyone start emuelec on this device?

Read the wiki, specifically the Getting Started page.

themegak, 2020-12-14:
I have a solution for this.  Not sure if this is mentioned anywhere else but here is how i solved the issue.  I also purchased the transpeed s905x3 4gb ram unit from Amazon, had the same issues everyone else mentioned.  I tried other files in the device tree and for the heck of it i tried the 2gb ram versions of  the s905x3 files.  This worked/solved my problem and EmuElec booted.  I can only assume that these off brand s905x3 4gb devices are really 2gb versions disguised as 4gb units.  Anyone who has issues booting EmuElec on s905x3 4gb units should try using the corresponding 2gb files in the device tree folder.  This should resolve your issue.  I want to bad mouth the sellers of these devices but isn't there Honor among thieves ?  Not sure, anywho hope this helps someone.

dm2912, 2020-12-14:
will always be the way with these generic boxes.. anything like t96, x86, max, pro, air, etc are all just generic cobblejobs from a variety of manufacturer's, that could have wildly different chips in, different wifi, reused nand from other devices, memory etc

Which is why when buying a box its always worth trying to get from a brand that is known to a degree. Beelink, Xiaomi, Tanix, Ugoos, Minix

ricardojlrufino, 2021-06-08:
I'm using dtb: meson-sm1-sei610.dtb to boot ARMBIAN. 
But same file not work to boot EMUELEC.

eddcv89, 2022-08-31:
hello everyone, the problem lies in the change of name to dtb, when in windows the supposed renaming of the file name and ext is done, it is not entirely correct, however in reality the file never undergoes a change of ext.
As an example I can tell you that in reality your file, despite changing the ext to .img, is actually called dtb.img.dtb, so you should enable seeing the file extension in the view tab of the windos explorer, this gives me I noticed when opening the usb in linux and realizing that it was wrongly renamed, this way this problem is corrected
Sorry for my translate

SmallUglyRabbit, 2023-02-02:
Bizzarely, I got it to work aswell. The problem is in the naming as the one above has posted. You have to rename 
sm1_s905x3_4g.dtb to img.dtb and put it in the root folder. 

My issue is I bought an even larger SD card and I'm trying to get it to run on the device and it doesn't seem to work even though the first one does. I used one of those small plastic swords to hit the reset button, the kind you get with cocktails... I think a piece may have broken off inside. Otherwise it works perfectly fine on the other card. Keep at it!