EmuELEC/EmuELEC #132: My Mecool M8S Pro L (LB) doesn't boot correctly anymore

varma86, 2020-03-10:
I installed 3.3 version (generic). It worked for a while.
Today I ran it and I found a boot issue. I will attach a video to let you see the problem.
Then I made a fresh installation of 3.4 version and the problem is still there.
The box works great with Android and CoreElec, so I tend to exclude an hardware issue.
How to fix it?
Thanks

https://photos.app.goo.gl/dQUF4fd3mcVZogmf7

shantigilbert, 2020-03-11:
> The dtb.img is the same used when it worked and the same as CoreElec.

if you mean the actual same dtb, please do not do this, use the ones included on each release,do not mix dtbs from other versions/distros, ever

varma86, 2020-03-12:
I tried every version (3.3 or 3.4) with its dtb and CoreElec 9.2.1 dtb.
I tried Lakka 2.3.1 with CoreElec dtb and Emuelec dtb, but it doesn't work at all. As before, boot splash image ok, then black screen.
I guess has something to do with resolution. Is there a config file where I can set manually the resolution?

shantigilbert, 2020-03-12:
https://github.com/EmuELEC/EmuELEC/wiki/Common-Questions

varma86, 2020-03-12:
I already tried using the file EE_VIDEO_MODE and writing 1080p. Is the correct syntax? Or should I have written something about HDMI?
With only 1080p didn't work.

shantigilbert, 2020-03-13:
1080p is the default so it makes no sense to put it on the file, I do not know why the blacks creen

varma86, 2020-03-13:
I solved setting 2160p60hz. But maybe the problem could be investigated. Could I help? Sending lots, if you guide me.
My TV is an Hitachi.

shantigilbert, 2020-03-15:
I do not have a 4k TV so I cannot investigate this, this seems to be a problem with your TV not accepting 1080p input for whatever reason.

closed 2020-03-15 by shantigilbert