EmuELEC/EmuELEC #135: Feature Requests

funkboy27, 2020-03-14:
I was wondering if it would be possible to add the following features:

**USB and SD card simultaneously.**

  I  noticed that when I plugged in my USB hard drive, emulation station then ignored the ROMS folder on the SD card.  It seems to be an either/or kinda deal currently.

**A menu option to change the order of the systems displayed.**

I attempted to do this by editing the es_systems.cfg in /storage/.config/emulationstation but my changes there weren't reflected on emulation station after a reboot.

Also, I noticed there was another es_systems.cfg in usr/config/emulationstation  but that is in a read-only part of the FS of course.

It seems that the menu is referencing the file in the USR folder if im not mistaken.

Thanks for all your hard work!

shantigilbert, 2020-03-14:
"USB and SD card simultaneously"

This is imposible

closed 2020-03-14 by shantigilbert

funkboy27, 2020-03-14:
> "USB and SD card simultaneously"
> 
> This is imposible

This is done with retropie on the pi.  I realize it’s different platforms of course.   Figured since it was done there, it could be done here.  Anyway, thanks again for a great app!

shantigilbert, 2020-03-14:
I have no idea how they can point 2 folders to one path... I don't think that is possible anywhere.

funkboy27, 2020-03-14:
> I have no idea how they can point 2 folders to one path... I don't think that is possible anywhere.

I believe it’s because it changes the roms folder to another name on the SD card to “localroms” and allows the roms folder to be referenced from the HDD.

Since I’m not familiar with the inner working of retropie, I can only speculate, BUT I think what’s happening is they’re allowing emulationstation to use both folders as an option for finding ROM folders.

I see that your software changes the SD folder to be called “roms2” (or so it seems).  Couldn’t you have ES check both folders for roms?

shantigilbert, 2020-03-15:
No, es only looks for one folder

asakous, 2020-03-31:
@shantigilbert  
could be overlayfs?

SumavisionQ5, 2020-03-31:
changing the order of the systems displayed is very easy, what you need to do is:
1.sort your platforms as you want in es_systems.cfg.
2.disable "sort collections and systems" in "game collection settings".
3.reboot.