EmuELEC/EmuELEC #137: Wifi & BT not work on my tv box x10 Max+ with amlogic s905x3

Exia-Ti885, 2020-03-15:
Hello, thank you for your awesome works, emuelec work great on my tv box x10 Max+ with amlogic s905x3, the problem is it not detect the wifi and BT, it use BT 4.1, for wifi chip I dont know how to check it, its just say on the box with dual 2.4-5Ghz,

If it possible to fix it? Thank you very much!

howels, 2020-03-15:
I had to use the manual pairing instructions (https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#prior-to-v30-alternate-method) but was able to connect a selection of Bluetooth gamepads.

Exia-Ti885, 2020-03-16:
> I had to use the manual pairing instructions (https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#prior-to-v30-alternate-method) but was able to connect a selection of Bluetooth gamepads.

Thank you I will try it

howels, 2020-03-23:
Please close if resolved @Exia-Ti885

BaltasarParreira, 2020-03-23:
@Exia-Ti885  My s905x3 box internal BT was not detected, I had to use an extra BT dongle on a USB port and pairing my PS3 controller manually like says on that link worked perfectly.

lbernstone, 2020-03-27:
Please post on the CoreELEC board to get this fixed. https://discourse.coreelec.org/t/looking-for-users-with-non-working-wifi-or-bt/

Exia-Ti885, 2020-03-28:
Thank you guys, I now can connect my BT controller with BT dongle, but it now auto pairing my controller, but I have to connect my controller to usb hub and pair manually on ES setting menu everytime I reboot my sistem, is there any way to turn on auto scanning BT when turn on my system?

BaltasarParreira, 2020-03-29:
@Exia-Ti885 My saves the configuration and next boot I just need to press the PS3 home button on the screen that shows saying that a controller is not connected.

Are you sure that you are using the right commands, like this:

To setup Bluetooth controller:
```
bluetoothctl 

default-agent
power on
discoverable on
pairable on
scan on
```
Then after that you turn on the controller discover button (home on a PS3) so you start seeing some MAC address's passing with words like new, etc. take note of yours and use it with the next commands:
```
trust ###### -> (Your MAC address)
connect  ###### -> (Your MAC Address)
```

Only after you get a sucefully connecte message i'ts connected, sometimes ask something and you have to type "Yes".

After successfully connection just type:
`exit`

After this I don't need to pair it any more, always works on next boot.

Exia-Ti885, 2020-03-29:
> @Exia-Ti885 Mine saves the configuration and on next boot I just need to press the PS3 home button on the screen that shows saying that a controller is not connected from Emulationstation.
> 
> Are you sure that you are using the right commands, like this:
> 
> To setup Bluetooth controller:
> 
> ```
> bluetoothctl 
> 
> default-agent
> power on
> discoverable on
> pairable on
> scan on
> ```
> 
> Then after that you turn on the controller discover button (home on a PS3) so you start seeing some MAC address's passing with words like new, etc. take note of yours and use it with the next commands:
> 
> ```
> trust ###### -> (Your MAC address)
> connect  ###### -> (Your MAC address)
> ```
> 
> Only after you get a sucefully connecte message i'ts connected, sometimes ask something and you have to type "Yes".
> 
> After successfully connection just type:
> `exit`
> 
> After this I don't need to pair it any more, always works on next boot.

Thank you very much! I will try this 👍

howels, 2020-03-29:
> Thank you very much! I will try this 👍

PS3 controllers are a special case, for PS3 controller the process is automatic:
1. Connect PS3 controller to USB port
2. LEDs on controller flash. Press center "playstation button" & wait a second
3. Disconnect USB, LEDs all flash
4. Press center playstation button again.
5. After a couple of seconds you should see only a single LED is lit, controller is connected via Bluetooth

In step 2 the controller is detected by emuelec and the Mac of emuelec's bt adapter is written into the controller, so that in 4 the controller can connect straight away without the scan&pair process. PS3 controllers are different from normal gamepads in the pairing phase.

Exia-Ti885, 2020-04-02:
Thank you very much guys, all is solved now, I close this thread 👍👍

closed 2020-04-02 by Exia-Ti885