EmuELEC/EmuELEC #138: Game load, but crashes when trying to start

Felower, 2020-03-15:
When trying to run a game, it shows like is working to start, but once "starting", it crashes, even pre-installed games won't run, only bomberman but no doom or quake for example, Thank you

howels, 2020-03-15:
Did you read the instructions? Look in the wiki and load the non-free files into roms/ports as the instructions show. 
https://github.com/EmuELEC/EmuELEC/wiki/Supported-Platforms-And--Correct-Rom-Path

Felower, 2020-03-15:
Thank you, will try that, but curios why the pre-installed games are not working, they are on the right path i guess, can it be issue with the resolution? I have a 27in monitor, thank you again for your help

shantigilbert, 2020-03-15:
There are not pre-installed games, only prince of Persia

closed 2020-03-15 by shantigilbert