EmuELEC/EmuELEC #140: How to connect a Bluetooth headset or speaker

zifeiwonet, 2020-03-18:
Bluetooth speaker that can be successfully connected through the Bluetooth controller settings，The menu interface sounds normal, Bluetooth will be disconnected when entering the game，How to connect the Bluetooth speaker stably？

zifeiwonet, 2020-04-17:
Why hasn't there been any reply

shantigilbert, 2020-05-07:
because maybe nobody knows? I do not use a BT speaker, BT is enabled for game-pads not sure about speakers.

coach1988, 2020-05-24:
I was wondering about the same,as I was at first using EmuELEC as an add-on in coreelec,where it worked up to the point where I started a game.i think in coreelec it is handled by pulseaudio,but I think RA isn't compiled to make use of that (if EmuELEC ships it at all and doesn't use alsa only).

shantigilbert, 2021-06-07:
I don't think this is possible, if anyone finds a way please let me know so I can reopen the issue.

closed 2021-06-07 by shantigilbert