EmuELEC/EmuELEC #141: T95 Max+ Not Booting into EMUELEC

Caninusouza, 2020-03-19:
Bought the exact same box recommended by ETA Prime on his video and followed his tutorial step by step multiple times. It keeps booting into Android TV. Tried other different files from the tree than the one he recommended and nothing. Any chance you may know what is happening?

Thank you

shantigilbert, 2020-03-19:
https://github.com/EmuELEC/EmuELEC/issues/129

closed 2020-03-19 by shantigilbert

snakect77, 2020-04-24:
I found the same problem with the same box t95max + amlogic s905x3, using apk aida64 mi from model tvbox franklin device franklin platform
can anyone start emuelec on this device?

DKAngel2000, 2020-04-24:
> I found the same problem with the same box t95max + amlogic s905x3, using apk aida64 mi from model tvbox franklin device franklin platform
> can anyone start emuelec on this device?

download terminal emulator, load it up and type reboot update with your sd card in and presuming you set the card up properly it will boot, as per the instructions above